Linutronix/elbe #25: elbe chg_archive deletes xml comments

richardcochran, 2015-07-09:
Running 'elbe chg_archive' does not preserve xml comments but rather deletes
them without any kind of warning.  This command really should be polite and leave
the comments alone when updating the <archive> section.  After all, somebody's
important notes or other critical information may be in the comment blocks.

Thanks,
Richard

manut, 2015-12-24:
This is caused, by loading the XML in our parser, adding the archive/changing the archive tag and let the
parser write it back to disk. (The parser ignores comments.) On a first look, i can't find any settings for the
parser to keep the comments. Maybe it's better to use sth like 'sed' here? :-/

cmhe, 2016-01-11:
Another option would be introducing a "comment" tag with arbitrary content. But that would be a bit hackish.

manut, 2017-06-07:
this is fixed in elbe 1.9.x

closed 2017-06-07 by manut