Linutronix/elbe #32: elbe initvm create bin-cdrom.iso misses some packages

manut, 2015-11-23:
elbe initvm create examples/bbb-jessie.xml
....
elbe initvm create elbe-build-*/bin-cdrom.iso
...
- package libmnl0 is not installed
- package libnetfilter-acct1 is not installed
- package libsigc++-2.0-0c2a is not installed
- package nfacct is not installed

jogness, 2017-10-17:
With elbe 2.1 this command now appears totally broken. I have a jessie-based XML file. I created the bin-cdrom.iso using elbe 2.1. Then tried to recreate everything, including initvm with:

    $ elbe initvm create bin-cdrom.iso

It got as far as booting the kernel, but then aborted because it could not find the kernel modules.

    [    1.658574] sr0: scsi3-mmc drive: 4x/4x cd/rw xa/form2 tray
    [    1.662041] cdrom: Uniform CD-ROM driver Revision: 3.20
    [    1.667408] sr 1:0:0:0: Attached scsi generic sg0 type 5
    Starting system log daemon: syslogd, klogd.
    mount: mounting /dev/fd0 on /media failed: No such file or directory
    umount: can't umount /media: Invalid argument
    mount: mounting /dev/fd0 on /media failed: No such file or directory
    Detecting hardware to find CD-ROM drives  ... 95%... 100%
    Scanning CD-ROM  ... 11%... 20%... 30%... 41%... 51%... 60%... 72%... 81%... 90%... 100%
    Load installer components from CD
    ---------------------------------
    No kernel modules were found. This probably is due to a mismatch between the
    kernel used by this version of the installer and the kernel version available
    in the archive.
    If you're installing from a mirror, you can work around this problem by
    choosing to install a different version of Debian. The install will probably
    fail to work if you continue without kernel modules.
    Continue the install without loading kernel modules?
      1: Yes      2: No [*]
    Prompt: '?' for help, default=2> 

There are no .udeb files on the bin-cdrom.iso. Is that the problem?

manut, 2017-11-06:
The .udeb files need to be on the bin-cdrom.iso - Yes this is the problem. It appears as soon as debian updates the installer and they cannot be downloaded from the debian archive anymore.

in 'elbepack/repomanager.py: gen_repo_conf()' is the reprepro config that should ensure that the udebs are on the cdrom.

Can you please debug this??

manut, 2018-04-27:
should be fixed meanwhile, if not feel free to reopen the issue.

jogness, 2018-04-28:
I am not sure why you think this should be fixed. A quick test shows the following packages are missing from the initvm created from the bin-cdrom.iso:
- libmnl0:amd64
- libnetfilter-acct1:amd64
- libsigc++-2.0-0c2a:amd64
- nfacct

When comparing the original target with the target created from the bin-cdrom.iso, the following packages were missing:
- gnupg
- gnupg-agent
- libassuan0:amd64
- libksba8:amd64
- libnpth0:amd64
- libsqlite3-0:amd64
- pinentry-curses

Also, I did a test where debian-security was included (both for initvm and target). In that case the security versions were installed in the initvm from bin-cdrom.iso. But for the target, the security packages were **not** installed. Instead, the versions from the primary mirror were installed. (The validation file correctly pointed this out.)

Looking at the bin-cdrom.iso contents, I see that security packages are in the initvm repo. But in the target repo there are only packages from the primary mirror.

jogness, 2018-04-28:
A correction from my previous message...

The initvm created from bin-cdrom.iso has the same problem with missing security packages as the target.

torbenh3, 2019-07-04:
most of these issues are fixed now.
However, since a cdrom build has noauth enabled, debootstrap does not install gpg and
so gpg does not end up in the build in any case.

need to specify gpg as extra package to debootstrap.

commit a76b76fa (2019-08-09): version bump v10

  * make "elbe check_updates" work again
  * check_updates: add option to build changelogxml
  * fix pbuilder for <noauth> case
  * debianize kernel: fail on dts failure (Fixes #163)
  * efilesystem: fix unicode exceptions in den_licence (Fixes: #133)
  * Protect 'ln' command in finetuning from excursion
  * debianize kernel: make firmware depending on kernel's version
    (Fixes: #173)
  * fix gpg key unlock timeout
  * filesystem: fix mkdir method for softlinks
  * Add options for building ISO image
  * Add gnupg2 to debootstrap in preseed (Fixes: #32)
  * Revert "repomanag [... cut]
closed 2019-08-09 by torbenh3