Linutronix/elbe #52: big E

bene42, 2016-01-12:
$ elbe initvm submit castellum.xml 
elbe control create_project failed.
Traceback (most recent call last):
  File "/usr/bin/elbe", line 61, in <module>
    cmdmod.run_command( sys.argv[2:] )
  File "/usr/lib/python2.7/dist-packages/elbepack/commands/control.py", line 102, in run_command
    action.execute (control, opt, args[1:])
  File "/usr/lib/python2.7/dist-packages/elbepack/soapclient.py", line 171, in execute
    print (client.service.create_project ( xml_base64 ))
  File "/usr/lib/python2.7/dist-packages/suds/client.py", line 542, in __call__
    return client.invoke(args, kwargs)
  File "/usr/lib/python2.7/dist-packages/suds/client.py", line 602, in invoke
    result = self.send(soapenv)
  File "/usr/lib/python2.7/dist-packages/suds/client.py", line 641, in send
    reply = transport.send(request)
  File "/usr/lib/python2.7/dist-packages/suds/transport/https.py", line 64, in send
    return  HttpTransport.send(self, request)
  File "/usr/lib/python2.7/dist-packages/suds/transport/http.py", line 77, in send
    fp = self.u2open(u2request)
  File "/usr/lib/python2.7/dist-packages/suds/transport/http.py", line 118, in u2open
    return url.open(u2request, timeout=tm)
  File "/usr/lib/python2.7/urllib2.py", line 431, in open
    response = self._open(req, data)
  File "/usr/lib/python2.7/urllib2.py", line 449, in _open
    '_open', req)
  File "/usr/lib/python2.7/urllib2.py", line 409, in _call_chain
    result = func(*args)
  File "/usr/lib/python2.7/urllib2.py", line 1227, in http_open
    return self.do_open(httplib.HTTPConnection, req)
  File "/usr/lib/python2.7/urllib2.py", line 1197, in do_open
    raise URLError(err)
urllib2.URLError: <urlopen error [Errno 104] Connection reset by peer>

Giving up
$

manut, 2016-02-04:
we should split all uploads into junks, like done in elbepack/soapclient.py:SetPdebuilderAction

can we introduce an upload function in the soap interface?! instead of duplicating this code in several actions?!

manut, 2017-01-03:
fixed with 98b98ab609de856af3c8b8015191ad04e97c105c

closed 2017-01-03 by manut