Linutronix/elbe #53: processing large XML file (66M) failed on second run

bene42, 2016-01-12:
Processing a large XML file failed on a second run:
$ elbe initvm submit castellum.xml
.... (build OK)
$ elbe initvm submit castellum.xml 
elbe control create_project failed.
Server returned error:

Elbe XML Validation Error
XML Parse error
Memory allocation failed : growing buffer

Giving up
$

After restarting the initvm the elbe initvm submit works as expected for the next run.
After increasing the memory of the initvm from 1024M to 2048M the project could be build several times.

manut, 2016-02-04:
maybe we should split away the archive tag before XML validation, or at least the error message should be improved

manut, 2017-01-03:
fixed with 98b98ab609de856af3c8b8015191ad04e97c105c

closed 2017-01-03 by manut