Linutronix/elbe #55: declaring <suite>stable</suite> failed to build

bene42, 2016-02-02:
Trying to build an image with <suite>stable</suite> failed silently:
```
...
Package List validation
-----------------------

No Errors found

Getting generated Files

Saving generated Files to elbe-build-20160202-181513
source.xml      (Current source.xml of the project)
cdrom.cpio      (Image)
licence.txt     (License file)
validation.txt  (Package list validation result)
elbe-report.txt         (Report)
log.txt         (Log file)
$
```
Looking into log.txt shows a build failure:
```
------------------------------------------------------------------------------

Build failed
Traceback (most recent call last):
  File "/usr/lib/python2.7/dist-packages/elbepack/asyncworker.py", line 86, in execute
    self.project.build(skip_pkglist=True, build_bin=self.build_bin, build_sources=self.build_src)
  File "/usr/lib/python2.7/dist-packages/elbepack/elbeproject.py", line 260, in build
    cdrom_size=cdrom_size )
  File "/usr/lib/python2.7/dist-packages/elbepack/cdroms.py", line 93, in mk_binary_cdrom
    repo_path, log, cdrom_size, mirror)
  File "/usr/lib/python2.7/dist-packages/elbepack/repomanager.py", line 213, in __init__
    maxsize )
  File "/usr/lib/python2.7/dist-packages/elbepack/repomanager.py", line 73, in __init__
    self.gen_repo_conf()
  File "/usr/lib/python2.7/dist-packages/elbepack/repomanager.py", line 96, in gen_repo_conf
    fp.write( "Suite: " + codename2suite[ att.codename ] + "\n" )
KeyError: 'stable'
```

manut, 2017-08-09:
i don't want elbe to work with 'stable' as suite. One should use 'stretch' instead. Otherwise the build result after 4 years will be completely different..

But this should be verified with the xml schema instead of producing a backtrace.

jogness, 2017-09-11:
I do not think it makes sense to verify it with the XML schema. That would require adding and maintaining the codenames in the XML schema. I think a simple code check against valid names would be enough and allow a useful message. Perhaps something similar to the buildtype check in ElbeDefaults.__init__().

jogness, 2017-09-11:
I just noticed issue#102, which is related to my last comment.
[https://github.com/Linutronix/elbe/issues/102](url)

bene42, 2017-09-11:
I see no difference in maintaining a list of valid names in some elbe source code files or in the XML schema. I would prefer the XML schema. XML schema is a well known and supported feature by many 
tools handling XML files. It allows tooling around E.L.B.E. to validate the receipt instead of handling E.L.B.E. error reports or copy'n'paste code.

commit a76b76fa (2019-08-09): version bump v10

  * make "elbe check_updates" work again
  * check_updates: add option to build changelogxml
  * fix pbuilder for <noauth> case
  * debianize kernel: fail on dts failure (Fixes #163)
  * efilesystem: fix unicode exceptions in den_licence (Fixes: #133)
  * Protect 'ln' command in finetuning from excursion
  * debianize kernel: make firmware depending on kernel's version
    (Fixes: #173)
  * fix gpg key unlock timeout
  * filesystem: fix mkdir method for softlinks
  * Add options for building ISO image
  * Add gnupg2 to debootstrap in preseed (Fixes: #32)
  * Revert "repomanag [... cut]
closed 2019-08-09 by torbenh3