Linutronix/elbe #60: Copy a file from rfs to the build directory

iluminat23, 2016-02-26:
It would be nice to be able to copy a file from the rfs to the build directory.

jogness, 2016-02-26:
By "build directory" do you mean "project directory"? i.e. the parent of the chroot filesystem? I am curious why this would be nice. What is the use case?

iluminat23, 2016-02-26:
I mean the directory where the images are saved (elbe-build-YYYYMMDD-hhmmss/). We create a rootfs.tar.gz which contains all files. We can extract for example the bootloader an the kernel image from there. But it would be nice if we could tell elbe to copy them to the build directory. So these files could be used directly.

manut, 2016-03-07:
yes, this would be really useful.
I think it should be a move from the /target directory instead of a copy.
A new finetuning action might be used to implement this..

iluminat23, 2016-03-14:
I think both would be nice to have. In my case I would need a copy, as the file should also stay in the image.

manut, 2018-12-21:
fixed by c931f0eed

closed 2018-12-21 by manut