Linutronix/elbe #63: there is no sane errorhandling if target image is to small for content

manut, 2016-04-12:
running cmd +cp -a "/var/cache/elbe/c2235b79-a071-4857-9d75-13167b6bb48c/filesystems/rfs"/* "/var/cache/elbe/c2235b79-a071-4857-9d75-13167b6bb48c/imagemnt"+
------------------------------------------------------------------------------
cp: error writing '/var/cache/elbe/c2235b79-a071-4857-9d75-13167b6bb48c/imagemnt/var/cache/apt/srcpkgcache.bin': No space left on device
cp: failed to extend '/var/cache/elbe/c2235b79-a071-4857-9d75-13167b6bb48c/imagemnt/var/cache/apt/srcpkgcache.bin': No space left on device
cp: cannot create directory '/var/cache/elbe/c2235b79-a071-4857-9d75-13167b6bb48c/imagemnt/var/cache/apt/archives': No space left on device
cp: cannot create directory '/var/cache/elbe/c2235b79-a071-4857-9d75-13167b6bb48c/imagemnt/var/cache/debconf': No space left on device
cp: cannot create directory '/var/cache/elbe/c2235b79-a071-4857-9d75-13167b6bb48c/imagemnt/var/cache/ldconfig': No space left on device
cp: cannot create directory '/var/cache/elbe/c2235b79-a071-4857-9d75-13167b6bb48c/imagemnt/var/cache/man': No space left on device
cp: cannot create directory '/var/cache/elbe/c2235b79-a071-4857-9d75-13167b6bb48c/imagemnt/var/cache/elbe': No space left on device
cp: cannot create directory '/var/cache/elbe/c2235b79-a071-4857-9d75-13167b6bb48c/imagemnt/var/tmp': No space left on device
cp: cannot create directory '/var/cache/elbe/c2235b79-a071-4857-9d75-13167b6bb48c/imagemnt/var/spool': No space left on device
cp: cannot create directory '/var/cache/elbe/c2235b79-a071-4857-9d75-13167b6bb48c/imagemnt/var/log': No space left on device
cp: cannot create directory '/var/cache/elbe/c2235b79-a071-4857-9d75-13167b6bb48c/imagemnt/var/local': No space left on device
cp: cannot create directory '/var/cache/elbe/c2235b79-a071-4857-9d75-13167b6bb48c/imagemnt/var/backups': No space left on device
cp: cannot create directory '/var/cache/elbe/c2235b79-a071-4857-9d75-13167b6bb48c/imagemnt/var/opt': No space left on device
cp: cannot create directory '/var/cache/elbe/c2235b79-a071-4857-9d75-13167b6bb48c/imagemnt/var/mail': No space left on device
------------------------------------------------------------------------------

manut, 2018-04-27:
this should be fixed together with #146

bene42, 2023-10-25:
Need info. Please reopen if needed.

closed 2023-10-25 by bene42