Linutronix/elbe #72: 'elbe control list_users' fails with a backtrace

AndreySV, 2016-06-18:
Here is the backtrace

$ elbe control list_users
Traceback (most recent call last):
  File "/usr/bin/elbe", line 61, in <module>
    cmdmod.run_command( sys.argv[2:] )
  File "/usr/lib/python2.7/dist-packages/elbepack/commands/control.py", line 102, in run_command
    action.execute (control, opt, args[1:])
  File "/usr/lib/python2.7/dist-packages/elbepack/soapclient.py", line 138, in execute
    users = client.service.list_users (client)
  File "/usr/lib/python2.7/dist-packages/suds/client.py", line 566, in __call__
    return client.invoke(args, kwargs)
  File "/usr/lib/python2.7/dist-packages/suds/client.py", line 700, in invoke
    soapenv = binding.get_message(self.method, args, kwargs)
  File "/usr/lib/python2.7/dist-packages/suds/bindings/binding.py", line 122, in get_message
    content = self.bodycontent(method, args, kwargs)
  File "/usr/lib/python2.7/dist-packages/suds/bindings/document.py", line 96, in bodycontent
    add_param, self.options().extraArgumentErrors)
  File "/usr/lib/python2.7/dist-packages/suds/argparser.py", line 83, in parse_args
    return arg_parser(args, kwargs, extra_parameter_errors)
  File "/usr/lib/python2.7/dist-packages/suds/argparser.py", line 109, in __call__
    return self.__all_parameters_processed()
  File "/usr/lib/python2.7/dist-packages/suds/argparser.py", line 146, in __all_parameters_processed
    self.__check_for_extra_arguments(args_required, args_allowed)
  File "/usr/lib/python2.7/dist-packages/suds/argparser.py", line 187, in __check_for_extra_arguments
    self.__error("".join(msg_parts))
  File "/usr/lib/python2.7/dist-packages/suds/argparser.py", line 196, in __error
    raise TypeError("%s() %s" % (self.__method_name, message))
TypeError: list_users() takes 0 positional arguments but 1 was given

manut, 2016-08-02:
are you sure, that the initvm also uses elbe 1.1 ?

manut, 2016-11-15:
no response and i can't reproduce this issue.

closed 2016-11-15 by manut