Linutronix/elbe #93: the `<rm>` finetuning rule is not working

iluminat23, 2017-02-14:
The `<rm>` finetuning rule doesn't work. I used the following rule:
`<rm>usr/lib/arm-linux-gnueabi/gconv/*</rm>`

But the content of the `gconv` directory was not removed from my image. When I use the following rule everything works fine:
`<command>rm -rvf usr/lib/arm-linux-gnueabi/gconv/*</command>`

richardcochran, 2017-02-17:
I can confirm this bug.  IIRC, this was working 1.9.15, but now in
1.9.17 it is broken.

It appears that the '*' is the problem. I have a line

    <rm>/var/cache/apt/archives/*.deb</rm>

that doesn't work, but the other rm command applied to single files or
directories does work.

Thanks,
Richard

AndreySV, 2017-03-05:
This problem was introduced by commit 6d778c5aa4ed69d3721809d86ca13952c033d7bd
According to commit message this is known downside of this change.

> commit 6d778c5aa4ed69d3721809d86ca13952c033d7bd (HEAD, refs/remotes/origin/devel/elbe-2.0, refs/remotes/elbe2/devel/elbe-2.0)
Author: Torben Hohn <torben.hohn@linutronix.de>
Date:   Mon Feb 20 12:37:06 2017 +0100

    Revert "finetuning: be aware of spaces in filenames for mv,cp,.."
    
    This reverts commit 657a609d4d097b264df8ce187510c5cb9eefba01.
    
    The Commit break wildcards in <rm>var/cache/apt/*</rm> etc.
    We need to come up with a better solution here.

manut, 2017-06-14:
fixed by 542b8f13c5f8c1a9af3ab35cb9627128512c2b97

closed 2017-06-14 by manut