Linutronix/elbe #98: dbsfed.xsd uses <import>

torbenh3, 2017-04-06:
```
  <import namespace="http://www.w3.org/XML/1998/namespace"
         schemaLocation="http://www.w3.org/2001/03/xml.xsd"/>

```
this results in an attempt to access the internet.

Possible workaround is specifying --skip-validate.

mischelebuha, 2017-04-19:
I have the same problem because I am sitting behind a company proxy.
How can I submit the --skip-validate flag to the "elbe initvm submit xxx.xml" command?

manut, 2017-04-26:
sometimes the include fails, because w3.org ist no accessible. We need to cache the files in a local dictionary

goldsimon, 2017-04-28:
As Mischelebuha wrote, this is not only about failed internet access but also about missing proxy settings. I have made "elbe initvm submit" work by hacking db.py, but that's clearly not a good solution. I want validation to work behind a proxy, too :-) (And don't ask me why we have a proxy, I wish we wouldn't!)

manut, 2017-06-07:
a proper fix is in work, see elbe-devel ML

manut, 2017-06-14:
fixed by d881f064eeaec5d4c19507c0ed99791c7c3ea0e0

closed 2017-06-14 by manut

goldsimon, 2017-06-14:
I'm still having trouble using elbe behind a proxy. Are there any additional steps required to make it use the locally cached XSDs?

manut, 2017-07-28:
no.

please open a new issue containing an error log. thx.