Linutronix/elbe #103: keys shall not be urls, but they should reside in xml files

torbenh3, 2017-06-07:


manut, 2018-12-21:
Being able to use keys from keyservers would be also nice

lzrdkng, 2019-05-20:
I'm working on this.

As for now, I added the following:

<raw-key>
----- BEGIN PGP ... -----
...
----- END PGP ... -----

</raw-key>

and

<key-server pool="eu.pool.sks-keyservers.net">0x36AA35FF22BB8F84</key-server>

Is this a good format?

lzrdkng, 2019-05-20:
An other format could be to make a complex type like 'pgp-key' and assign to it some attribute:

```
<key type="raw">
----- BEGIN PGP ... -----
...
----- END PGP ... -----

</key>
````

`<key type="server" pool="eu.pool.sks-keyservers.net">0x36AA35FF22BB8F84</key>`

The default type would be to "url" to maintain compatibility

bgermann, 2019-05-20:
The pool attribute should contain the protocol (hkp://, hkps://, ...) that is used for the key transfer.

lzrdkng, 2019-05-20:
> The pool attribute should contain the protocol (hkp://, hkps://, ...) that is used for the key transfer.

From GPG(1):

> The format of the name is a URI: `scheme:[//]keyservername[:port]' The scheme is the type of keyserver: "hkp" for the HTTP (or compat‐
>               ible) keyservers, "ldap" for the LDAP keyservers, or "mailto" for the Graff email keyserver. Note that your particular installation of GnuPG  may  have  other
>               keyserver types available as well. Keyserver schemes are case-insensitive.

This will be documented in the XML

torbenh3, 2019-05-29:
merged into elbe master now.
keyserver download is not implemented.

open a new issue for it, if you are interested in it.

closed 2019-05-29 by torbenh3