Linutronix/elbe #104: add key section to the mirror tag

mischelebuha, 2017-06-30:
A key tag should be added to the mirror tag in the initvm and project section.

If a local Debian repository is used, which is either not complete or is simply self-signed, the transfer of a key must be possible.

In the url list this was also implemented.

    <mirror>
        <primary_host>localhost123</primary_host>
        <primary_path>debian</primary_path>
        <primary_proto>http</primary_proto>
        <pirmary_key>http://localhost123/my.pub</primary_key>
        <url-list>
            <url>
                <binary>http://localhost123/debian jessie main</binary>
                <source>http://localhost123/debian jessie main</source>
                <key>http://localhost123/my.pub</key>
            </url>
        </url-list>
    </mirror>

torbenh3, 2019-07-04:
when a primary_key is specified, we should also refrain from validating against the default debian keys.
hope that debootsrap allows to configure the key somehow.

bgermann, 2021-03-08:
This was implemented a long time ago. There is now another option to specify raw-key, which takes the base64-encoded PEM.

kempfsn, 2023-03-05:
raw-key does not solve this, see #365.