Linutronix/elbe #109: add control over the finetuning process

lwalewski, 2017-07-10:
Add support for multiple <finetuning> sections that would be executed on different stages of the RFS build, e.g.:

`<finetuning runafter='imagegen'>`

The new algorithm should also take into account relationships between the `<finetuning>` and the overlay. Extracting the archive is done twice at the moment. But this should also be controllable with an additional attribute (`before_ft, after_ft`) if not given: old behavior.

Motivation:
We are planning to use the cryptsetup package to encrypt the RFS generated with elbe. One way to do that is to mount the RFS as a loop device and manipulate it using cryptsetup. Since mounting a loop device and manipulating it requires root privileges we would like to encapsulate those tasks. The elbe initvm provides a perfect, secure environment for such tasks and we are using it anyway so there would be no additional cost in setting things up.

manut, 2018-01-09:
2nd extract of archive is removed by 502442c7b42eb7a385e2f932e5dddba48015b43c

manut, 2018-12-21:
A project-finteuning rule was added, usage can bee seen here: 8d8071c72a1834e95a2dff1e8553931e234c2290

Now there could be a project-finetuning rule for cryptsetup. Can you make a suggestion?

commit 7425de56 (2019-10-02): debian changelog: bump version to v12.0

  * initvm is now using buster (debian 10)
  * logging has been completely reimplemented, and uses
    the python logging framework. (Fixes #109)
  * Errors now result in build_failures, and build_failure results
    in error return codes, and no files to download. This is especially
    the case for finetuning errors which have been ignored before.
    (Fixes: #208, #146, #131, #127, #144, #70)
  * honour ELBE_SOAPPORT and ELBE_SSHPORT in "elbe initvm create"
  * src-cdrom: add size limit in XML
  * init: replace localhost with 10.0.2.2 in initvm proxy [... cut]
closed 2019-10-02 by torbenh3