Linutronix/elbe #110: "initvm create" doesn't work on i386 architecture

lwalewski, 2017-07-24:
Wrong architecture for the `elbe-bootstrap` package is selected, when `elbe initvm create` is issued on an i386 machine, i.e. amd64 is selected instead of i386:

```
$ lscpu
Architecture:          i686
CPU op-mode(s):        32-bit
[...]

$ elbe initvm create
[...]
Reading package lists... Done
Building dependency tree... Done
Reading package lists... Done
Building dependency tree... Done
--2017-07-24 22:17:29--  http://debian.linutronix.de/elbe-common/pool/main/e/elbe-bootstrap/elbe-bootstrap_20150422+deb8u4_amd64.deb
Resolving debian.linutronix.de (debian.linutronix.de)... 146.0.238.70
Connecting to debian.linutronix.de (debian.linutronix.de)|146.0.238.70|:80... connected.
HTTP request sent, awaiting response... 200 OK
Length: 24546432 (23M) [application/x-debian-package]
Saving to: '/tmp/tmphM2XOG/pkg.deb'

/tmp/tmphM2XOG/pkg.deb   100%[==================================>]  23.41M   405KB/s    in 50s     

2017-07-24 22:18:19 (484 KB/s) - '/tmp/tmphM2XOG/pkg.deb' saved [24546432/24546432]
```

After invoking /usr/bin/kvm from the Makefile the boot process hangs with the following error:

```
SeaBIOS (version 1.10.2-1)

iPXE (http://ipxe.org) 00:03.0 CA00 PCI 2.10 PNP PMM+07F90BD0+07ED0BD0 CA00

Booting from ROM...
This kernel requires an x86-64 CPU, but only detected an i686 CPU.
Unable to boot - please use a kernel appropriate for your CPU.
```

The i386 version of the elbe-bootstrap package is missing in the elbe-common repository.

It is likely that the same defect is present in all but amd64 architectures.

lwalewski, 2017-07-24:
The same behaviour if you copy the `default-init.xml` file, change the buildtype to i386:

```<buildtype>i386</buildtype>```

and then run `initvm create` using the modified XML file:

```
$ elbe initvm create i386-init.xml
```

manut, 2017-07-27:
elbe-bootstrap is only available for amd64 architecture. It should also be downloaded on other architectures. That is working correctly.

You need to set the interpreter to qemu-system-x86_64 to boot the 64-bit kernel from the elbe-bootstrap package. But this will be slow as hell!

We recommend to use a x86_64 machine for using elbe.

closed 2017-07-27 by manut