Linutronix/elbe #113: elbe 2 initvm broken

richardcochran, 2017-08-11:
In elbe 2, the default XML and the examples

    /usr/lib/python2.7/dist-packages/elbepack/init/default-init.xml
    /usr/share/doc/elbe-doc/examples/elbe-init-with-ssh.xml

all still have

    <binary>http://debian.linutronix.de/elbe-testing jessie main</binary>

left over from 1.9.x.  As a result, when you create a new VM using

    elbe initvm create

the result is not usable, since

    elbe initvm submit myfile.xml

throws the following error.

    elbe control create_project failed.
    elbe v2.9.1 is used in initvm, this is not compatible with elbe v2.0 that is used on this machine. Please install same versions of elbe in initvm and on your machine.

richardcochran, 2017-08-11:
Workaround:  Copy and edit a suitable XML file, for example

    /usr/lib/python2.7/dist-packages/elbepack/init/default-init.xml
    /usr/share/doc/elbe-doc/examples/elbe-init-with-ssh.xml

and replace

    http://debian.linutronix.de/elbe-testing

with

    http://debian.linutronix.de/elbe

chslion, 2017-08-11:
_for jessie user on vmware_
>  pc-i440fx-2.6

be aware that qemu has no 2.6 version of the initvm machine. so you might want to edit the Makefile.mako aswell

manut, 2017-08-14:
fixed in f971494fff175c08b7468def4373ab9d6f17d80e and elbe-2.1

@chslion please make an extra issue for this. Can you provide me a Machine name, that works for all debian releases on vmware?

closed 2017-08-14 by manut