Linutronix/elbe #114: qemu hardcoded machine pc-i440fx-2.6 not found in debian jessie on vmware

chslion, 2017-08-15:
elbe wants to start pc-i440fx-2.6, but this Machine is not installed in qemu..

**Docker Environment**

> root@4b619dcd715b:/# qemu-system-x86_64 -machine help
Supported machines are:
pc-0.13              Standard PC (i440FX + PIIX, 1996)
pc-i440fx-2.0        Standard PC (i440FX + PIIX, 1996)
pc-1.0               Standard PC (i440FX + PIIX, 1996)
pc                   Standard PC (i440FX + PIIX, 1996) (alias of pc-i440fx-2.1)
pc-i440fx-2.1        Standard PC (i440FX + PIIX, 1996) (default)
pc-q35-1.7           Standard PC (Q35 + ICH9, 2009)
pc-1.1               Standard PC (i440FX + PIIX, 1996)
pc-0.14              Standard PC (i440FX + PIIX, 1996)
pc-q35-2.0           Standard PC (Q35 + ICH9, 2009)
pc-i440fx-1.4        Standard PC (i440FX + PIIX, 1996)
pc-i440fx-1.5        Standard PC (i440FX + PIIX, 1996)
pc-0.15              Standard PC (i440FX + PIIX, 1996)
xenfv                Xen Fully-virtualized PC
pc-q35-1.4           Standard PC (Q35 + ICH9, 2009)
isapc                ISA-only PC
pc-1.2               Standard PC (i440FX + PIIX, 1996)
pc-0.10              Standard PC (i440FX + PIIX, 1996)
pc-0.11              Standard PC (i440FX + PIIX, 1996)
q35                  Standard PC (Q35 + ICH9, 2009) (alias of pc-q35-2.1)
pc-q35-2.1           Standard PC (Q35 + ICH9, 2009)
pc-i440fx-1.7        Standard PC (i440FX + PIIX, 1996)
pc-i440fx-1.6        Standard PC (i440FX + PIIX, 1996)
none                 empty machine
xenpv                Xen Para-virtualized PC
pc-q35-1.5           Standard PC (Q35 + ICH9, 2009)
pc-q35-1.6           Standard PC (Q35 + ICH9, 2009)
pc-0.12              Standard PC (i440FX + PIIX, 1996)
pc-1.3               Standard PC (i440FX + PIIX, 1996)
QEMU emulator version 2.1.2 (Debian 1:2.1+dfsg-12+deb8u6), Copyright (c) 2003-2008 Fabrice Bellard

**Linux Mint (Debian 8) Environment**

> qemu-system-x86_64 -machine help
Supported machines are:
ubuntu               Ubuntu 16.04 PC (i440FX + PIIX, 1996) (alias of pc-i440fx-xenial)
pc-i440fx-xenial     Ubuntu 16.04 PC (i440FX + PIIX, 1996) (default)
pc-i440fx-wily       Ubuntu 15.04 PC (i440FX + PIIX, 1996)
pc-i440fx-vivid      Ubuntu 15.04 PC (i440FX + PIIX, 1996)
pc-i440fx-utopic     Ubuntu 14.10 PC (i440FX + PIIX, 1996)
pc-i440fx-trusty     Ubuntu 14.04 PC (i440FX + PIIX, 1996)
pc                   Standard PC (i440FX + PIIX, 1996) (alias of pc-i440fx-2.5)
pc-i440fx-2.5        Standard PC (i440FX + PIIX, 1996)
pc-i440fx-2.4        Standard PC (i440FX + PIIX, 1996)
pc-i440fx-2.3        Standard PC (i440FX + PIIX, 1996)
pc-i440fx-2.2        Standard PC (i440FX + PIIX, 1996)
pc-i440fx-2.1        Standard PC (i440FX + PIIX, 1996)
pc-i440fx-2.0        Standard PC (i440FX + PIIX, 1996)
pc-i440fx-1.7        Standard PC (i440FX + PIIX, 1996)
pc-i440fx-1.6        Standard PC (i440FX + PIIX, 1996)
pc-i440fx-1.5        Standard PC (i440FX + PIIX, 1996)
pc-i440fx-1.4        Standard PC (i440FX + PIIX, 1996)
pc-1.3               Standard PC (i440FX + PIIX, 1996)
pc-1.2               Standard PC (i440FX + PIIX, 1996)
pc-1.1               Standard PC (i440FX + PIIX, 1996)
pc-1.0               Standard PC (i440FX + PIIX, 1996)
pc-0.15              Standard PC (i440FX + PIIX, 1996)
pc-0.14              Standard PC (i440FX + PIIX, 1996)
pc-0.13              Standard PC (i440FX + PIIX, 1996)
pc-0.12              Standard PC (i440FX + PIIX, 1996)
pc-0.11              Standard PC (i440FX + PIIX, 1996)
pc-0.10              Standard PC (i440FX + PIIX, 1996)
pc-q35-xenial        Ubuntu 16.04 PC (Q35 + ICH9, 2009)
q35                  Standard PC (Q35 + ICH9, 2009) (alias of pc-q35-2.5)
pc-q35-2.5           Standard PC (Q35 + ICH9, 2009)
pc-q35-2.4           Standard PC (Q35 + ICH9, 2009)
pc-q35-2.3           Standard PC (Q35 + ICH9, 2009)
pc-q35-2.2           Standard PC (Q35 + ICH9, 2009)
pc-q35-2.1           Standard PC (Q35 + ICH9, 2009)
pc-q35-2.0           Standard PC (Q35 + ICH9, 2009)
pc-q35-1.7           Standard PC (Q35 + ICH9, 2009)
pc-q35-1.6           Standard PC (Q35 + ICH9, 2009)
pc-q35-1.5           Standard PC (Q35 + ICH9, 2009)
pc-q35-1.4           Standard PC (Q35 + ICH9, 2009)
isapc                ISA-only PC
none                 empty machine
xenfv                Xen Fully-virtualized PC
xenpv                Xen Para-virtualized PC

> qemu-system-x86_64 --version
QEMU emulator version 2.5.0 (Debian 1:2.5+dfsg-5ubuntu10.14), Copyright (c) 2003-2008 Fabrice Bellard

quickfix is to override MACHINE to pc which defaults to latest installed machine version.

`elbe init && make MACHINE=pc ...`

manut, 2017-08-15:
pc-i440fx-2.6 should only be used if 'systemd-detect-virt' returns vmware. There was a bug report, that this workaround is needed to make 'elbe initvm start/attach' work on vmware based hosts. If 'pc' is used in this case 'elbe initvm attach' will never display a login prompt.

manut, 2018-01-09:
fixed by e46272dd8ec4990a79bb9b1cf4a82d86b519a975

closed 2018-01-09 by manut