Linutronix/elbe #116: We need a possibility to add custom HTML-Elements to the XMLs

mischelebuha, 2017-08-23:
It should be possible to add custom HTML elements to the ELBE XML.

Background: Each XML represents a device. We have planned to program a tool that will notify certain developers as soon as there are DSA's for packages that are referenced in an XML. It would be nice if we could add an e-mail list to the XML. Then only the developers responsible for this device are notified.

manut, 2017-08-23:
Please don't program an extra tool for that. We have an elbe subcommand called 'check_updates' (https://github.com/Linutronix/elbe/blob/master/elbepack/commands/check_updates.py). Currently it prints a list of updateable packages to stdout. Please add the ability for sending mails to this subcommand. Then you can add a grammer for sth. like this:
<maintainers-list><maintainer><name>My Name</name><mail>my@name.domain</mail></name></maintainer></maintainer-list>
to our schema (https://github.com/Linutronix/elbe/blob/master/schema/dbsfed.xsd) and read the mail adresses in 'elbe check_updates' from this list.

manut, 2018-04-27:
instead of coding the adresses into XML, it might be nicer to have a cmdline parameter at check_updates to specify the adresses, smtpserver, etc.

bgermann, 2022-08-24:
As this bug is open for a long time and nobody has provided any input, I am closing this.
It is always possible to add additional XML elements in their own namespace, which should be the best option in this case.
Then you can process the information in your own tool.