Linutronix/elbe #118: Cannot boot BBB image build by elbe

nandra, 2017-09-05:
I've build test image for BBB but it won't boot. If I insepct /boot partition of SDcard I've seen there is kernel+initrd version 4.9.3 but dtb was from 3.16 + also uEnv.txt has also 3.16 records? Any ideas? I've installed elbe from linutronix repo directly. Thanks.

manut, 2017-09-05:
can you provide some bootlog? Or don't you even get an uboot prompt?

nandra, 2017-09-07:
I did some small hacking on BBB to have it even booted but still no luck to boot to rootfs:
```
U-Boot SPL 2016.11+dfsg1-4 (Mar 27 2017 - 18:39:51)
Trying to boot from MMC1
MMC partition switch failed
*** Warning - MMC partition switch failed, using default environment

reading u-boot.img
reading u-boot.img

U-Boot 2016.11+dfsg1-4 (Mar 27 2017 - 18:39:51 +0000)

CPU  : AM335X-GP rev 2.1
I2C:   ready
DRAM:  512 MiB
MMC:   OMAP SD/MMC: 0, OMAP SD/MMC: 1
*** Warning - bad CRC, using default environment

<ethaddr> not set. Validating first E-fuse MAC
Net:   cpsw
Press SPACE to abort autoboot in 2 seconds
switch to partitions #0, OK
mmc0 is current device
SD/MMC found on device 0
reading boot.scr
** Unable to read file boot.scr **
reading uEnv.txt
256 bytes read in 5 ms (49.8 KiB/s)
Loaded env from uEnv.txt
Importing environment from mmc0 ...
Running uenvcmd ...
reading vmlinuz-4.9.0-3-armmp
3701544 bytes read in 240 ms (14.7 MiB/s)
reading am335x-boneblack.dtb
34184 bytes read in 13 ms (2.5 MiB/s)
reading initrd.img-4.9.0-3-armmp
16023296 bytes read in 1011 ms (15.1 MiB/s)
## Flattened Device Tree blob at 82000000
   Booting using the fdt blob at 0x82000000
   Loading Ramdisk to 90000000, end 90000000 ... OK
   Loading Device Tree to 8fff4000, end 8ffff587 ... OK

Starting kernel ...

[    0.000000] Booting Linux on physical CPU 0x0
[    0.000000] Linux version 4.9.0-3-armmp (debian-kernel@lists.debian.org) (gcc version 6.3.0 20170516 (Debian 6.3.0-18) ) #1 SMP Debian 4.9.30-2+deb9u2 (2017-06-26)
[    0.000000] CPU: ARMv7 Processor [413fc082] revision 2 (ARMv7), cr=10c5387d
[    0.000000] CPU: PIPT / VIPT nonaliasing data cache, VIPT aliasing instruction cache
[    0.000000] OF: fdt:Machine model: TI AM335x BeagleBone Black
[    0.000000] efi: Getting EFI parameters from FDT:
[    0.000000] efi: UEFI not found.
[    0.000000] cma: Reserved 16 MiB at 0x9e800000
[    0.000000] Memory policy: Data cache writeback
[    0.000000] CPU: All CPU(s) started in SVC mode.
[    0.000000] AM335X ES2.1 (sgx neon)
[    0.000000] percpu: Embedded 14 pages/cpu @df946000 s27660 r8192 d21492 u57344
[    0.000000] Built 1 zonelists in Zone order, mobility grouping on.  Total pages: 129408
[    0.000000] Kernel command line: console=ttyO0,115200 root=/dev/mmcblk0p2 rootwait
[    0.000000] PID hash table entries: 2048 (order: 1, 8192 bytes)
[    0.000000] Dentry cache hash table entries: 65536 (order: 6, 262144 bytes)
[    0.000000] Inode-cache hash table entries: 32768 (order: 5, 131072 bytes)
[    0.000000] Memory: 488012K/522240K available (7168K kernel code, 957K rwdata, 2208K rodata, 1024K init, 333K bss, 17844K reserved, 16384K cma-reserved, 0K highmem)
[    0.000000] Virtual kernel memory layout:
[    0.000000]     vector  : 0xffff0000 - 0xffff1000   (   4 kB)
[    0.000000]     fixmap  : 0xffc00000 - 0xfff00000   (3072 kB)
[    0.000000]     vmalloc : 0xe0800000 - 0xff800000   ( 496 MB)
[    0.000000]     lowmem  : 0xc0000000 - 0xe0000000   ( 512 MB)
[    0.000000]     pkmap   : 0xbfe00000 - 0xc0000000   (   2 MB)
[    0.000000]     modules : 0xbf000000 - 0xbfe00000   (  14 MB)
[    0.000000]       .text : 0xc0008000 - 0xc0800000   (8160 kB)
[... 82 lines cut ...]
[    1.048140] ledtrig-cpu: registered to indicate activity on CPUs
[    1.055437] NET: Registered protocol family 10
[    1.061402] mip6: Mobile IPv6
[    1.064643] NET: Registered protocol family 17
[    1.069329] mpls_gso: MPLS GSO support
[    1.073383] omap_voltage_late_init: Voltage driver support not added
[    1.080073] sr_dev_init: No voltage domain specified for smartreflex0. Cannot initialize
[    1.088548] sr_dev_init: No voltage domain specified for smartreflex1. Cannot initialize
[    1.097282] ThumbEE CPU extension supported.
[    1.101795] Registering SWP/SWPB emulation handler
[    1.106827] SmartReflex Class3 initialized
[    1.112067] registered taskstats version 1
[    1.116499] zswap: loaded using pool lzo/zbud
[    1.121302] ima: No TPM chip found, activating TPM-bypass!
[    1.133948] omap_i2c 44e0b000.i2c: bus 0 rev0.11 at 400 kHz
[    1.141785] omap_i2c 4819c000.i2c: bus 2 rev0.11 at 100 kHz
[    1.148234] omap_rtc 44e3e000.rtc: setting system clock to 2000-01-01 00:00:00 UTC (946684800)
[    1.157310] sr_init: No PMIC hook to init smartreflex
[    1.162804] sr_init: platform driver register failed for SR
[    1.169101] vmmcsd_fixed: disabling
[    1.173345] Waiting for root device /dev/mmcblk0p2...
[    9.644225] random: fast init done
```

seems that SDcard is not mouted. What I did is to update dtb to be found in uEnv script.

manut, 2017-09-07:
looks good so far. Can you provide the change needed for the uEnv.txt ?

Have you tried another sdcard vendor? The card seems to be not even detected by the kernel.
Can you exchange rootwait by rootdelay=10
This avoids waiting indefinitely for the RFS and printing an error instead containing the available disks and partitions.

nandra, 2017-09-07:
I've changed uEnv.txt to following:
```uenvcmd=setenv bootargs 'console=ttyO0,115200 root=/dev/mmcblk0p2 rootwait';load mmc 0:1 0x84000000 vmlinuz-4.9.0-3-armmp;load mmc 0:1 0x82000000 am335x-boneblack.dtb;load mmc 0:1 0x88000000 initrd.img-4.9.0-3-armmp;bootz 0x84000000 0x88000000: 0x82000000
```

and copy am335x-boneblack.dtb from deb package. As I already mention there si a problem that SD card is not detected and as you propose with rootdelay:
```
[    1.174177] Waiting 10 sec before mounting root device...
[    2.835628] random: fast init done
[   11.424854] List of all partitions:
[   11.428509] No filesystem could mount root, tried: [   11.433486] 
[   11.435076] Kernel panic - not syncing: VFS: Unable to mount root fs on unknown-block(0,0)
[   11.443718] CPU: 0 PID: 1 Comm: swapper/0 Not tainted 4.9.0-3-armmp #1 Debian 4.9.30-2+deb9u2
[   11.452621] Hardware name: Generic AM33XX (Flattened Device Tree)
[   11.459031] [<c0115178>] (unwind_backtrace) from [<c010f324>] (show_stack+0x20/0x24)
[   11.467131] [<c010f324>] (show_stack) from [<c040e27c>] (dump_stack+0x90/0xa4)
[   11.474693] [<c040e27c>] (dump_stack) from [<c026d100>] (panic+0x100/0x28c)
[   11.481982] [<c026d100>] (panic) from [<c0b01530>] (mount_block_root+0x2b4/0x30c)
[   11.489807] [<c0b01530>] (mount_block_root) from [<c0b01604>] (mount_root+0x7c/0x80)
[   11.497902] [<c0b01604>] (mount_root) from [<c0b01760>] (prepare_namespace+0x158/0x1a0)
[   11.506268] [<c0b01760>] (prepare_namespace) from [<c0b0107c>] (kernel_init_freeable+0x274/0x288)
[   11.515549] [<c0b0107c>] (kernel_init_freeable) from [<c07f3070>] (kernel_init+0x18/0x124)
[   11.524199] [<c07f3070>] (kernel_init) from [<c010a2f8>] (ret_from_fork+0x14/0x3c)
[   11.532125] ---[ end Kernel panic - not syncing: VFS: Unable to mount root fs on unknown-block(0,0)
```

I'll try different SDcard though

nandra, 2017-09-07:
Same issue with other SDcard

manut, 2017-09-13:
[   11.424854] List of all partitions:

as you see, the list is empty. So the kernel didn't detect your SDCard.
This is either a problem with your hardware / sdcard or a problem of the debian kernel/devicetree.
This issue can't be solved with elbe :(

manut, 2017-09-13:
However we should check if another hw, sdcard is working and if not, switch the example to a HW really supported by debian.

djfun, 2017-09-22:
The 4.9.0-3-armmp kernel from debian also fails to read from the sdcard on the beaglebone black which I have here. A 4.9.47 kernel compiled with omap2plus_defconfig however does work.
The debian kernel might be missing some config needed for accessing the sdcard. `CONFIG_MMC_OMAP_HS`?

djfun, 2017-09-25:
After some more testing I got it to work: My elbe xml did some cleaning of /boot which accidentally removed the initrd file. It is now included in my image and I made sure u-boot looks at the correct location for it.
Comparing my boot log to the one posted above, the earliest difference I could find was that mine has `[    0.348881] Unpacking initramfs...` and `[    1.431125] Freeing initrd memory: 15648K (cf0b8000 - d0000000)` which I don't see in the log here.

nandra, 2017-09-25:
Can you please share your findings (patch or PR). Thanks

PR #124 (2017-09-28): Fix BBB example
the finetuning command for writing the uenvcmd did substitute the ${filesize} variable (for an empty string). Because of this, the kernel did not properly read the initrd and was unable to read the rootfs on the sdcard.
should fix #118
closed 2017-09-28 by manut