Linutronix/elbe #129: xsdtoasciidoc skips at least some attributes

manut, 2017-10-27:
<simpleContent>
      <extension base="rfs:string">
        <attribute name="pin" type="string" use="optional">
          <annotation>
            <documentation>
              Pin-Priority of packages from this source.
            </documentation>

Is not available here:

https://elbe-rfs.org/docs/sphinx/article-elbe-schema-reference.html

manut, 2017-11-06:
also e.g. the finetuning tag <command> is missing

torbenh3, 2018-05-24:
complex types which are not sequences do not get examples generated for them.

torbenh3, 2018-05-24:
notes on the original issue:

we iterate over all xsd nodes. but we only have handlers for:

 * `ELEM = "{http://www.w3.org/2001/XMLSchema}element"`
 * `CPLX = "{http://www.w3.org/2001/XMLSchema}complexType"`
 * `SIMP = "{http://www.w3.org/2001/XMLSchema}simpleType"`

commit a76b76fa (2019-08-09): version bump v10

  * make "elbe check_updates" work again
  * check_updates: add option to build changelogxml
  * fix pbuilder for <noauth> case
  * debianize kernel: fail on dts failure (Fixes #163)
  * efilesystem: fix unicode exceptions in den_licence (Fixes: #133)
  * Protect 'ln' command in finetuning from excursion
  * debianize kernel: make firmware depending on kernel's version
    (Fixes: #173)
  * fix gpg key unlock timeout
  * filesystem: fix mkdir method for softlinks
  * Add options for building ISO image
  * Add gnupg2 to debootstrap in preseed (Fixes: #32)
  * Revert "repomanag [... cut]
closed 2019-08-09 by torbenh3