Linutronix/elbe #131: elbe should be strict, if source can not be found

torbenh3, 2017-12-06:
currently elbe silently ignores sources, that can not be found.
This is wrong.

However, it still must be possible to install packages, without source.
Normally customers app is such a package, where source is not wanted on the src-cdrom.iso

these packages should be marked somehow. <pkg closed_source="1"> or something.

this would also allow a customer repo to contain the sourcecode, which will not hit src-cdrom

torbenh3, 2017-12-06:
just found on_src_cd, this allows us, to strictly make it an error, if source is not found.

oldiob, 2019-07-22:
I don't really understand the problem here.  Nothing is silence, i.e. that if the source is not found, an error message is logged.  See cdrom.py:73

Do you want an exception to be raise instead?

closed 2020-03-16 by bgermann