Linutronix/elbe #132: race condition with env vars

torbenh3, 2017-12-06:
env vars a global in process.
We have to reset http_proxy and friends, when they are empty.

This points at a race condition, though.

Identify where they are used, and either use execve to set them.
Or make sure that the right vars come to RpcAptCache.

virraaj, 2022-07-18:
Can you please give some more information to reproduce the issue? and can you please verify if the issue is still present in elbe v14.4?

bgermann, 2022-08-17:
On 3rd Dec 2018, @manut wrote: "patches are on the ML but not tested so far"
@virraaj Please check if you can identify the patches on [elbe-devel](https://lists.linutronix.de/pipermail/elbe-devel/)

virraaj, 2022-08-17:
@bgermann the question here is about reproducing. I don't have enough information on what to test even if I guess any patches.

bgermann, 2022-08-17:
The patches will probably provide more info for reproducing.

bene42, 2023-10-25:
More information needed. Please reopen if nessesary.

closed 2023-10-25 by bene42