Linutronix/elbe #133: care about encoding in licence.txt/xml generation

manut, 2017-12-08:
But the current code raises an exception that interrupts the hole build.
This means, if you use a package that triggers an encoding exception you are
not able to build the image, that's a nogo.

I'm quite unhappy with the licence code. It's hard to understand and
undocumented, also the encoding issue needs to be resolved.

Also it is not py3 compatible

manut, 2018-01-05:
according to https://www.debian.org/doc/debian-policy/#copyright-information all copyright files must be encoded in utf-8. 

So elbe can expect that and if handling the file fails, we should handle this gently and recommend to file a bug for the package that cause the exception.

torbenh3, 2019-07-04:
there is an error attribute, that defaults to strict, when an encoding error is seen.
use errors="replace" and remove the exception handling then.

commit a76b76fa (2019-08-09): version bump v10

  * make "elbe check_updates" work again
  * check_updates: add option to build changelogxml
  * fix pbuilder for <noauth> case
  * debianize kernel: fail on dts failure (Fixes #163)
  * efilesystem: fix unicode exceptions in den_licence (Fixes: #133)
  * Protect 'ln' command in finetuning from excursion
  * debianize kernel: make firmware depending on kernel's version
    (Fixes: #173)
  * fix gpg key unlock timeout
  * filesystem: fix mkdir method for softlinks
  * Add options for building ISO image
  * Add gnupg2 to debootstrap in preseed (Fixes: #32)
  * Revert "repomanag [... cut]
closed 2019-08-09 by torbenh3