Linutronix/elbe #136: remove python 2 compatibility code

manut, 2017-12-19:
as soon as the transition to py3 is completed the following commits need to be reverted

c1addeeb7ac3c6a7e3b46a28e9e8bca2ee444fdd
py3: debian - produce python3 in python2 packages

189cd630e7347f374724dcf0aac1afc4f2ae657b
py3: debian - override tests

efc7b37b66ace6a3c6014bfdd17c8c0dc3d49af5
py3: make - use python3 interpreter to call xsdtoasciidoc

c6a19f1dbb1e3c83b46b5139831f6a6b6714ada4
add an iter() implementation to treeutils

24ac676d06bd8d8dd3808af126102b8a28f85f91
py3: enforce elbe to run with python2

manut, 2018-01-15:
07e1b93795221b095db51e364d47d3cb7c6eef36
py3: debian/python-buildenv-postinst: call elbe with py2

manut, 2018-04-26:
one blocker is: https://bugs.debian.org/cgi-bin/bugreport.cgi?bug=877783

manut, 2018-04-26:
https://github.com/arskom/spyne/issues/113

bgermann, 2019-09-02:
arskom/spyne#617

closed 2020-09-23 by bgermann