Linutronix/elbe #141: I can start an InitVM built with the wrong elbe Version

waynix, 2018-01-22:
elbe v2.2:

    waynix@my-elbe-vm:~$ elbe initvm start --directory=initvm_built_with_elbe_v2.1
    *****
    waynix@my-elbe-vm:~$ elbe initvm stop
    elbe v2.1 is used in initvm, this is not compatible with elbe v2.2 that is used on this machine. Please install same versions of elbe in initvm and on your machine.

waynix, 2018-01-22:
workaround to stop the initvm:
    waynix@my-elbe-vm:~$ elbe initvm attach OR tmux attach
    waynix@my-elbe-vm:~$ <login as root>
    waynix@my-elbe-vm:~$ poweroff

bgermann, 2021-03-08:
We do not have a property to find out the installed elbe on an initvm. Also, an initvm could be updated by means of apt. So, having a property that can be read by libvirt can become invalid and it is not trivial to update. So this will not be fixed.

closed 2021-03-08 by bgermann