Linutronix/elbe #143: elbe initvm create with xinclude returns error message

schlatterbeck, 2018-01-30:
I have an XML file where the <initvm> stanza is in an include file. When I issue
elbe initvm create /path/to/xml-file the initvm is successfully created and started (indicating that the xinclude *did* work?) and the I'm getting:

Requesting system reboot
[ 1033.450475] reboot: Restarting system
mkdir -p .stamps
touch .stamps/stamp-install-initial-image
*****
XML file is inavlid: /home/ralf/checkout/own/projects/sunxi-elbe/elbe-sunxi.xml:7 error Element '{http://www.w3.org/2003/XInclude}include': This element is not expected. Expected is one of ( initvm, project, target, archive, debootstrappkgs, initvmpkgs, initvm_sources_list, initvm_apt_prefs, fullpkgs, sources_list ).

zsh: exit 20    elbe initvm create ~/checkout/own/projects/sunxi-elbe/elbe-sunxi.xml

The xml file starts:
<ns0:RootFileSystem
 xmlns:ns0="https://www.linutronix.de/projects/Elbe"
 xmlns:xi="http://www.w3.org/2003/XInclude"
 xmlns:xsi="http://www.w3.org/2001/XMLSchema-instance"
 created="2009-05-20T08:50:56" revision="6"
 xsi:schemaLocation="https://www.linutronix.de/projects/Elbe dbsfed.xsd">
  <xi:include href="include/initvm.xml"/>
  <xi:include href="include/project.xml"/>
  <target>

Version is 
: ralf@bee:133; dpkg -l | grep elbe
ii  elbe                                  2.2+stretch251                        all          Embedded Linux Build Environment

If needed I can provide the full xml files.

manut, 2018-01-31:
elbe initvm create might work, because it checks if an initvm tag is available; if not it uses the default-initvm.xml from the 'elbepack'.

You need to you 'elbe preprocess' to 'remove' the xincludes before using any other elbe initvm/control/pbuilder command.

If this fixes your problem, do you have any suggestions howto imrpove the documentation.

schlatterbeck, 2018-01-31:
On Wed, Jan 31, 2018 at 09:25:33AM +0000, Manuel Traut wrote:
> elbe initvm create might work, because it checks if an initvm tag is
> available; if not it uses the default-initvm.xml from the 'elbepack'.
> 
> You need to you 'elbe preprocess' to 'remove' the xincludes before
> using any other elbe initvm/control/pbuilder command.

Hmm, can't this be changed to do this automagically? Why does the user
have to bother that the xml needs preprocessing?

> If this fixes your problem, do you have any suggestions howto imrpove
> the documentation.

I'd rather like to improve the processing of the xml to do this. It
looks to me like the other commands (like initvm submit) *do* the
preprocessing step?

Ralf
-- 
Dr. Ralf Schlatterbeck                  Tel:   +43/2243/26465-16
Open Source Consulting                  www:   http://www.runtux.com
Reichergasse 131, A-3411 Weidling       email: office@runtux.com

manut, 2018-01-31:
you're right. It is done in elbe-initvm-submit but not in create. This is a bug! There is already another issue mentioning the code duplication between those two commands: https://github.com/Linutronix/elbe/issues/84

bene42, 2023-10-25:
Not planed ATM.

closed 2023-10-25 by bene42