Linutronix/elbe #149: elbe-bootstrap download: dont use fallback code if python-apt can't verify package

manut, 2018-04-09:
if python-apt is used to download the elbe-bootstrap package, but the 'elbe repo key' is not installed. The package can not be verified. So it will not be used. But than the python-apt fallback code is used to download the package. This is wrong. Dont use the fallback code if python-apt is working. Print a message, that the package cant be used, because verfication failed and give a hint to install the elbe repo key.

jogness, 2018-04-09:
A better hint would be to add the elbe repo key as a "key" entry in the XML file. There is no reason that a user should have to modify local files in their system. Especially something as important as their keyring.

manut, 2018-04-26:
Since 478b26e08550f1322ec2cb07db26d8fa961054b3 an exception should occur. It still should be converted to an user readable error message

manut, 2018-04-26:
This is related to #145