Linutronix/elbe #152: docker container fails to build examples (missing module deb822)

torbjoernk, 2018-04-17:
Within the docker container (branch release/v2.9.9), it is not possible to build the elbe examples:
```elbe@08a747989b13:~$ /elbe/elbe initvm submit /elbe/examples/armhf-ti-beaglebone-black.xml 
elbe control create_project failed.
Traceback (most recent call last):
  File "/elbe/elbe", line 62, in <module>
    mod = __import__(modname)
  File "/elbe/elbepack/commands/control.py", line 29, in <module>
    from elbepack.soapclient import ClientAction, ElbeSoapClient
  File "/elbe/elbepack/soapclient.py", line 27, in <module>
    import deb822   # package for dealing with Debian related data
ImportError: No module named deb822

Giving up```

manut, 2018-04-18:
http://elbe-devel.2373613.n4.nabble.com/elbe-devel-PATCH-docker-add-python-debian-package-tt890.html

manut, 2018-04-26:
fixed in 12fd4f0

closed 2018-04-26 by manut