Linutronix/elbe #156: build examples/x86_64-pc-hdimg-readonly-rfs.xml fails

manut, 2018-05-03:
building examples/x86_64-pc-hdimg-readonly-rfs.xml fails with devel/elbe-3.0:

running cmd +chroot /var/cache/elbe/b3a4c2c8-7225-4706-9232-7e1fce447067/chroot /bin/sh -c '[ -f /lib/systemd/system/serial-getty@.service ] && ln -s /lib/systemd/system/serial-getty@.service /etc/systemd/system/getty.target.wants/serial-getty@ttyS0.service'+
------------------------------------------------------------------------------
------------------------------------------------------------------------------

Command failed with errorcode 1
Build failed
Traceback (most recent call last):
  File "/usr/lib/python2.7/dist-packages/elbepack/asyncworker.py", line 129, in execute
    skip_pbuild=self.skip_pbuilder)
  File "/usr/lib/python2.7/dist-packages/elbepack/elbeproject.py", line 489, in build
    self.targetfs.part_target(self.builddir, grub_version, grub_fw_type)
UnboundLocalError: local variable 'grub_fw_type' referenced before assignment

Exception:
local variable 'grub_fw_type' referenced before assignment

/main libkeyutils1 amd64 1.5.5-3+deb7u1 [8664 B]

torbenh3, 2018-05-04:
das muss wohl ein wheezy build sein. ist das richtig ?

diff --git a/elbepack/elbeproject.py b/elbepack/elbeproject.py
index 690a289d..59feffa6 100644
--- a/elbepack/elbeproject.py
+++ b/elbepack/elbeproject.py
@@ -470,6 +470,7 @@ class ElbeProject (object):
         elif self.get_rpcaptcache().is_installed('grub-pc'):
             if self.codename == "wheezy":
                 grub_version = 199
+                grub_fw_type = "???"
             else:
                 grub_version = 202
                 grub_fw_type = "bios"

manut, 2018-05-04:
fixed by 4f8ef75776118e4ceb0064bd84dc95c5b00f8dd2, thx Martin!

closed 2018-05-04 by manut