Linutronix/elbe #159: 'elbe initvm' behavior has changed (elbe-3.0)

haspv, 2018-05-09:
since elbe 3.0 the behavior of 'elbe initvm create' has changed.
If you want to create a second initvm, you have to set the environment variable ELBE_INITVM_DOMAIN
The default value is "initvm". You have to change the name to create a second initvm.

Is it possible to store this value in the initvm directory?
Otherwise you have to specifie the initvm directory an the environment variable to boot a specific initvm.

tested with version 2.9.10

manut, 2018-05-09:
> since elbe 3.0 the behavior of 'elbe initvm create' has changed.
> If you want to create a second initvm, you have to set the environment variable ELBE_INITVM_DOMAIN
> The default value is "initvm". You [...]

it should be stored in the xml file that is in this directory.

> Otherwise you have to specifie the initvm directory an the environment variable to boot a specific initvm.

i think the environment variable should be all that's needed.
directory should be just needed for creation.

maybe it would be good to have a --name parameter instead of the environment
variable.

to keep the old behaviour, we could read the name from the xml if --directory
was given and print a deprecated warning for a while?

manut, 2019-03-12:
fixed by d74d8b69912d4a790ee7892c22dbae78ae2643cb

closed 2019-03-12 by manut