Linutronix/elbe #161: wait_busy runs into tcp connection timeout

manut, 2018-05-17:
this is seen while big images are gzipped inside the initvm

manut, 2018-08-02:
This happened during initvm submit:

01:26:35.050 Thu Aug  2 10:38:12 2018 -- pmstatus:linux-image-4.9.0-7-armmp:91.7355:Configuring linux-image-4.9.0-7-armmp (armhf)
01:28:56.526 Traceback (most recent call last):
01:28:56.526   File "./elbe", line 55, in <module>
01:28:56.526     cmdmod.run_command(sys.argv[2:])
01:28:56.526   File "/elbe/elbepack/commands/control.py", line 161, in run_command
01:28:56.526     action.execute(control, opt, args[1:])
01:28:56.526   File "/elbe/elbepack/soapclient.py", line 570, in execute
01:28:56.527     busy = client.service.get_project_busy(builddir, part)
01:28:56.527   File "/usr/lib/python2.7/dist-packages/suds/client.py", line 566, in __call__
01:28:56.527     return client.invoke(args, kwargs)
01:28:56.527   File "/usr/lib/python2.7/dist-packages/suds/client.py", line 705, in invoke
01:28:56.527     result = self.send(soapenv)
01:28:56.527   File "/usr/lib/python2.7/dist-packages/suds/client.py", line 747, in send
01:28:56.527     reply = self.options.transport.send(request)
01:28:56.527   File "/usr/lib/python2.7/dist-packages/suds/transport/https.py", line 66, in send
01:28:56.527     return HttpTransport.send(self, request)
01:28:56.527   File "/usr/lib/python2.7/dist-packages/suds/transport/http.py", line 80, in send
01:28:56.527     fp = self.u2open(u2request)
01:28:56.527   File "/usr/lib/python2.7/dist-packages/suds/transport/http.py", line 127, in u2open
01:28:56.527     return url.open(u2request, timeout=tm)
01:28:56.527   File "/usr/lib/python2.7/urllib2.py", line 429, in open
01:28:56.527     response = self._open(req, data)
01:28:56.527   File "/usr/lib/python2.7/urllib2.py", line 447, in _open
01:28:56.527     '_open', req)
01:28:56.527   File "/usr/lib/python2.7/urllib2.py", line 407, in _call_chain
01:28:56.527     result = func(*args)
01:28:56.527   File "/usr/lib/python2.7/urllib2.py", line 1228, in http_open
01:28:56.527     return self.do_open(httplib.HTTPConnection, req)
01:28:56.527   File "/usr/lib/python2.7/urllib2.py", line 1201, in do_open
01:28:56.527     r = h.getresponse(buffering=True)
01:28:56.527   File "/usr/lib/python2.7/httplib.py", line 1121, in getresponse
01:28:56.527     response.begin()
01:28:56.527   File "/usr/lib/python2.7/httplib.py", line 438, in begin
01:28:56.527     version, status, reason = self._read_status()
01:28:56.527   File "/usr/lib/python2.7/httplib.py", line 394, in _read_status
01:28:56.527     line = self.fp.readline(_MAXLINE + 1)
01:28:56.527   File "/usr/lib/python2.7/socket.py", line 480, in readline
01:28:56.527     data = self._sock.recv(self._rbufsize)
01:28:56.527 socket.error: [Errno 104] Connection reset by peer

manut, 2018-08-03:
and a similar error, at a different build-state:

Fri Aug  3 08:15:22 2018 -- ------------------------------------------------------------------------------
Fri Aug  3 08:15:22 2018 -- Warning: database 'wheezy|main|source' was modified but no index file was exported.
Fri Aug  3 08:15:22 2018 -- Changes will only be visible after the next 'export'!
Fri Aug  3 08:15:22 2018 -- ------------------------------------------------------------------------------
Fri Aug  3 08:15:22 2018 --                                                     
Fri Aug  3 08:15:22 2018 -- running cmd +reprepro --keepunreferencedfiles --export=never --basedir "/var/cache/elbe/22578a05-8dc2-4f01-a2c6-052e348e3fc2/srcrepo" -C main -P normal -S misc includedsc wheezy /var/cache/elbe/sources/python-setuptools_5.5.1-1.dsc+
Traceback (most recent call last):                                              
  File "./elbe", line 55, in <module>                                           
    cmdmod.run_command(sys.argv[2:])                                            
  File "/home/elbe/elbepack/commands/control.py", line 161, in run_command      
    action.execute(control, opt, args[1:])                                      
  File "/home/elbe/elbepack/soapclient.py", line 570, in execute                
    busy = client.service.get_project_busy(builddir, part)                      
  File "/usr/lib/python2.7/dist-packages/suds/client.py", line 566, in __call__ 
    return client.invoke(args, kwargs)                                          
  File "/usr/lib/python2.7/dist-packages/suds/client.py", line 705, in invoke   
    result = self.send(soapenv)                                                 
  File "/usr/lib/python2.7/dist-packages/suds/client.py", line 747, in send     
    reply = self.options.transport.send(request)                                
  File "/usr/lib/python2.7/dist-packages/suds/transport/https.py", line 66, in send
    return HttpTransport.send(self, request)                                    
  File "/usr/lib/python2.7/dist-packages/suds/transport/http.py", line 80, in send
    fp = self.u2open(u2request)                                                 
  File "/usr/lib/python2.7/dist-packages/suds/transport/http.py", line 127, in u2open
    return url.open(u2request, timeout=tm)                                      
  File "/usr/lib/python2.7/urllib2.py", line 429, in open                       
    response = self._open(req, data)                                            
  File "/usr/lib/python2.7/urllib2.py", line 447, in _open                      
    '_open', req)                                                               
  File "/usr/lib/python2.7/urllib2.py", line 407, in _call_chain                
    result = func(*args)                                                        
  File "/usr/lib/python2.7/urllib2.py", line 1228, in http_open                 
    return self.do_open(httplib.HTTPConnection, req)                            
  File "/usr/lib/python2.7/urllib2.py", line 1201, in do_open                   
    r = h.getresponse(buffering=True)                                           
  File "/usr/lib/python2.7/httplib.py", line 1121, in getresponse               
    response.begin()                                                            
  File "/usr/lib/python2.7/httplib.py", line 438, in begin                      
    version, status, reason = self._read_status()                               
  File "/usr/lib/python2.7/httplib.py", line 394, in _read_status               
    line = self.fp.readline(_MAXLINE + 1)                                       
  File "/usr/lib/python2.7/socket.py", line 480, in readline                    
    data = self._sock.recv(self._rbufsize)                                      
socket.error: [Errno 104] Connection reset by peer

bene42, 2023-10-25:
More info needed. Reopen if nessesary.

closed 2023-10-25 by bene42