Linutronix/elbe #162: missing packets on source iso

haspv, 2018-05-17:
Not all source packets are on the iso if you build an stretch rfs with a jessie initvm.
A short investigation showed, that this happen, if initvm and target use the same packet in an different version.

manut, 2018-05-24:
can you give some details? E.g. target XML and what packages are missing?

bgermann, 2022-08-15:
Note on reproducing this: The key in the description is “A short investigation showed, that this happen, if initvm and target use the same packet in an different version.” This can maybe be reproduced by constructing an XML that installs some package in a target that is also installed in the initvm (but in a different version). One can build a booworm target which is bulleye (initvm) + 1 and choose a package which has changed its version from bullseye to bookworm.

virraaj, 2022-08-17:
Okay, for now on elbe 14.4, I am able to check the source package(bash) of different version on the target(buster), when initvm has a bash based on bullseye. So, I am not able to reproduce it. @bgermann do you have any more comments?

bgermann, 2022-08-17:
What exactly did you check? Is bash available in both source cdroms with the respective version?

virraaj, 2022-08-17:
I checked the log file for the build:
[CMD] reprepro --keepunreferencedfiles --export=never --basedir "/var/cache/elbe/75e19bd9-286d-4f12-aac5-cba09b36f2c4/srcrepo-target" --priority normal --section misc --component "target" includedsc buster /var/cache/elbe/75e19bd9-286d-4f12-aac5-cba09b36f2c4/chroot/var/cache/elbe/sources/bash_5.0-4.dsc
Warning: database 'buster|target|source' was modified but no index file was exported.
Changes will only be visible after the next 'export'!

I checked source-cdrom-target.iso by mounting it:
$ find . -name "bash*"
./pool/target/b/bash
./pool/target/b/bash/bash_5.0-4.debian.tar.xz
./pool/target/b/bash/bash_5.0-4.dsc
./pool/target/b/bash/bash_5.0.orig.tar.xz

And, I checked the bash installed in initvm by attaching it:
root@elbe-daemon:~# bash --version
GNU bash, version 5.1.4(1)-release (x86_64-pc-linux-gnu)
Copyright (C) 2020 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>

This is free software; you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.

bgermann, 2022-08-17:
Okay. You checked one version to be available in the target source cdrom. Is 5.1.4 available in src-cdrom-main.iso?

virraaj, 2022-08-17:
Okay, In src-cdrom-main.iso, i see:
$ find . -name "bash*"
./pool/initvm/b/bash
./pool/initvm/b/bash/bash_5.1-2+deb11u1.debian.tar.xz
./pool/initvm/b/bash/bash_5.1-2+deb11u1.dsc
./pool/initvm/b/bash/bash_5.1.orig.tar.xz
./pool/main/b/bash
./pool/main/b/bash/bash_5.0-4.debian.tar.xz
./pool/main/b/bash/bash_5.0-4.dsc
./pool/main/b/bash/bash_5.0.orig.tar.xz

bgermann, 2022-08-17:
That should be the right version. Thanks.

closed 2022-08-17 by bgermann