Linutronix/elbe #168: Is there a way to inject custom Debian packages directly without a repository server?

QuickJack, 2018-06-14:
I am currently adding these packages via the <archive> section and installing them from within the <finetuning> section using `<command>dpkg -i ...</command>`. This is working fine. However, the licensing information from these packages is not processed by Elbe.

manut, 2018-09-12:
yes, you can upload them to the project repo, the workflow looks sth. like this:

elbe control create_project # prints out project_dir
elbe control set_xml <project_dir> <xmlfile>
elbe prjrepo upload_pkg <project_dir> [<debfile> | <dscfile>]
elbe control build <project_dir>
elbe control wait_busy <project_dir>
elbe control --output elbe-build get_files <project_dir>

you can also post your questions at 'elbe-users@linutronix.de'

closed 2018-09-12 by manut

QuickJack, 2018-09-12:
OK, this method requires us to recompile the Debain packages. But I guess, it is impossible to inject binary Debian packages which have already been compiled before.

We are creating these packages within a well defined environment. All we need to do, is to inject these pre-built .deb files into the image created by Elbe.

manut, 2018-09-12:
No, this is exactly what you want.

The binary packages uploaded with upload_pkg are stored in a project local repository that is available as the packages from the pkg-list are installed. (It is not available during debootstraping the base-system)

If you want to build the packages from source, you need the 'elbe pbuilder' command.