Linutronix/elbe #174: elbe initvm create --skip-build sources fails if deb-src doesn't exist

manut, 2018-08-13:
Building an initvm with --skip-build-sources tries to access the deb-src mirror. However this is not needed for the desired build output.

-----
Reading package lists... Done
Building dependency tree... Done
E:Failed to fetch http://canoe/elbe-testing/dists/jessie/main/source/Sources  404  Not Found [IP: 10.100.31.140 80], E:Failed to fetch http://canoe/elbe-common/dists/jessie/main/source/Sources  404  Not Found [IP: 10.100.31.140 80], E:Failed to fetch http://canoe/debian/dists/jessie-backports/main/source/Sources  404  Not Found [IP: 10.100.31.140 80], E:Failed to fetch http://canoe//debian/dists/jessie/main/source/Sources  404  Not Found [IP: 10.100.31.140 80], E:Some index files failed to download. They have been ignored, or old ones used instead.
Reading package lists... Done
python-apt failed, using fallback code: 
---------------------------------------------------------------------------
Traceback (most recent call last):
  File "/usr/lib/python2.7/multiprocessing/managers.py", line 207, in handle_request
    result = func(c, *args, **kwds)
  File "/usr/lib/python2.7/multiprocessing/managers.py", line 386, in create
    obj = callable(*args, **kwds)
  File "/home/local/projects/elbe/elbe/elbepack/virtapt.py", line 113, in __init__
    self.cache = apt_pkg.Cache()
Error: E:The value 'jessie' is invalid for APT::Default-Release as such a release is not available in the sources
---------------------------------------------------------------------------
Failure to download kernel/initrd debian Package:

could not retreive nonexistent://elbe-bootstrap

Check Mirror configuration
'elbe init' Failed
Giving up

manut, 2018-12-03:
fixed by 47aaa2e6b0614d93ff532cc8846c2d3d7c956d87

torbenh3, 2019-07-04:
fixed... close it

closed 2019-07-04 by torbenh3