Linutronix/elbe #175: cdrom: rebuild is missing gpg

torbenh3, 2018-08-29:
when elbe is not building cross (iE amd64 or i386), and building from cdrom,
debootstrap will get the --no-check-gpg option. 

I am suspecting, that this stops it from installing gnupg.
In the End the rebuild is not the same, and Package Validation complains.

`ELBE Package validation
=======================

Package List validation
-----------------------

No Errors found

Full Packagelist validation
---------------------------

- package gnupg is not installed
- package gnupg-agent is not installed
- package libassuan0 is not installed
- package libksba8 is not installed
- package libnpth0 is not installed
- package pinentry-curses is not installed

Binary CD
==========

Source CD
==========

E:You must put some 'source' URIs in your sources.list

Getting generated Files
`

torbenh3, 2018-08-29:
`         if not self.xml.is_cross(host_arch):
            # ignore gpg verification if install from cdrom, cause debootstrap
            # seems to ignore /etc/apt/trusted.gpg.d/elbe-keyring.gpg
            # 01/2017 manut
            if self.xml.has(
                    "project/noauth") or self.xml.has("project/mirror/cdrom"):
                cmd = 'debootstrap --no-check-gpg --arch=%s "%s" "%s" "%s"' % (
                    arch, suite, self.rfs.path, primary_mirror)
            else:
                cmd = 'debootstrap --arch=%s "%s" "%s" "%s"' % (
                    arch, suite, self.rfs.path, primary_mirror)

            try:
                self.cdrom_mount()
                self.log.do(cmd)
            except CommandError:
                cleanup = True
                raise DebootstrapException()
            finally:
                self.cdrom_umount()
                if cleanup:
                    self.rfs.rmtree("/")

            return

        if self.xml.has("project/noauth"):
            cmd = 'debootstrap --no-check-gpg --foreign ' \
                  '--arch=%s "%s" "%s" "%s"' % (arch, suite, self.rfs.path,
                                                primary_mirror)
        else:
            if self.xml.has("project/mirror/cdrom"):
                keyring = ' --keyring="%s/targetrepo/elbe-keyring.gpg"' % (
                    self.rfs.fname("cdrom"))
            else:
                keyring = ''
            cmd = 'debootstrap --foreign --arch=%s %s "%s" "%s" "%s"' % (
                arch, keyring, suite, self.rfs.path, primary_mirror)

`

manut, 2019-04-16:
gpg is now always added to debootstrap --include: bb4d7e88e7c13ccd1042e16a298b4499b9c77223

closed 2019-04-16 by manut