Linutronix/elbe #176: elbe-daemon: crash during download_file

jogness, 2018-09-06:
After generating a large image (6GB) and then downloading the image gz, the elbe-daemon is sometimes crashing. On the client side I see:

Traceback (most recent call last):
  File "/usr/bin/elbe", line 55, in <module>
    cmdmod.run_command(sys.argv[2:])
  File "/usr/lib/python2.7/dist-packages/elbepack/commands/control.py", line 161, in run_command
    action.execute(control, opt, args[1:])
  File "/usr/lib/python2.7/dist-packages/elbepack/soapclient.py", line 445, in execute
    client.download_file(builddir, filename, dst_fname)
  File "/usr/lib/python2.7/dist-packages/elbepack/soapclient.py", line 88, in download_file
    ret = self.service.get_file(builddir, filename, part)
  File "/usr/lib/python2.7/dist-packages/suds/client.py", line 566, in __call__
    return client.invoke(args, kwargs)
  File "/usr/lib/python2.7/dist-packages/suds/client.py", line 705, in invoke
    result = self.send(soapenv)
  File "/usr/lib/python2.7/dist-packages/suds/client.py", line 753, in send
    return self.process_reply(reply.message, None, None)
  File "/usr/lib/python2.7/dist-packages/suds/client.py", line 805, in process_reply
    replyroot = _parse(reply)
  File "/usr/lib/python2.7/dist-packages/suds/client.py", line 947, in _parse
    return suds.sax.parser.Parser().parse(string=string)
  File "/usr/lib/python2.7/dist-packages/suds/sax/parser.py", line 131, in parse
    sax.parse(source)
  File "/usr/lib/python2.7/xml/sax/expatreader.py", line 110, in parse
    xmlreader.IncrementalParser.parse(self, source)
  File "/usr/lib/python2.7/xml/sax/xmlreader.py", line 125, in parse
    self.close()
  File "/usr/lib/python2.7/xml/sax/expatreader.py", line 225, in close
    self.feed("", isFinal = 1)
  File "/usr/lib/python2.7/xml/sax/expatreader.py", line 217, in feed
    self._err_handler.fatalError(exc)
  File "/usr/lib/python2.7/xml/sax/handler.py", line 38, in fatalError
    raise exception
xml.sax._exceptions.SAXParseException: <unknown>:2:29161: no element found

dschlegel, 2018-09-06:
I probably had the same problem when generating larger images. The issue was that the initvm ran out of RAM. 

You could try to increase the amount of memory the initvm gets. I don't remember the exact parameter but you can find it in the Makefile in the initvm directory (default value is 1G if I remember correctly). 

If you start the initvm using `$ _MEM_SIZE_PARAMETER_=10G elbe initvm start` it should work

jogness, 2018-09-06:
Thanks for the tip. This would imply that there is some sort of memory-leak in elbe-daemon. If I restart the daemon, the download always works. It is only when I generate an image and then download (without restarting elbe-daemon) that elbe-daemon crashes.

virraaj, 2022-07-18:
I have already worked on projects with image of 128GB compressed. It works at least on the Elbe that is based on python3. This stack trace is about python2. @bgermann , in my opinion, we can close this.

closed 2022-07-18 by bgermann