Linutronix/elbe #184: Large archive uploads fail sporadically

a-darwish, 2018-10-26:
ELBE v2.3

ELBE v2.3 is showing a Heisenbug behavior with large archives (100+ MiB), sometimes succeeding and sometimes producing below error (error is always the same):

```
$ elbe chg_archive .gen/devmachine.xml .gen/dev-archive.tar.bz2
$ elbe initvm submit --skip-build-bin --skip-build-source --output build .gen/devmachine.xml
elbe control set_xml failed2
Server returned error:
Elbe XML Validation Error
XML Parse error
xmlSAX2Characters, line 2775734, column 29
Giving up
Makefile:55: recipe for target 'dev-dev' failed
make: *** [dev-dev] Error 20
```

bgermann, 2019-03-05:
This is caused by a compilation-time libxml2 limit.

I can think of the following approaches to work around that:
1) Use iterparse() to parse the XML file and leave out the archive node from the tree, externalizing its (base64-decoded) content it in a temporary file and provide means to access it.
2) Change elbe not to use lxml. Instead use the built-in ElementTree. I checked it not to have a limit in place. Additionally, the schema would have to be validated via another library because that is the main lxml feature that is used by elbe. Or keep schema validation with lxml and advise the user to skip validation for large XML files.

bgermann, 2019-03-06:
So I think 04661fa3a329 and 3a1261f1f7 have fixed this issue. It should be closed. The 1.2 GiB limit still holds.

manut, 2019-03-11:
We don't want to support XML > 1.2 GiB. Large XML files don't make sense at all. The archive was designed to add some config files to the RFS.

I guess we should add an error in the XML validation if the file is larger than 1 GiB.

closed 2019-03-13 by manut

a-darwish, 2019-03-14:
Thanks @bgermann  and @manut for fixing this :+1: