Linutronix/elbe #185: Format of generated /etc/hosts file

QuickJack, 2018-10-30:
Given is the following xml fragment:
```
<target>
		<hostname>test</hostname>
		<domain>linutronix.de</domain>
(...)
</target>
```
The /etc/hosts file is generated as follows:

```
127.0.0.1	localhost
::1		localhost ip6-localhost ip6-loopback
ff02::1		ip6-allnodes
ff02::2		ip6-allrouters

127.0.0.1 test.linutronix.de test elbe-daemon
```

The format of the /etc/hosts file for Debian Stretch is described here: https://manpages.debian.org/stretch/manpages/hosts.5.en.html. According to this document, the hostname should be specified using 127.0.1.1 and not 127.0.01. There should be a tab instead of a space between 127.0.1.1 and the hostname.

When giving an empty **<domain></domain>** section, the **/etc/hosts** file will contain test.None. In most cases we don't need/want a domain entry. There seems to be no way to tell Elbe to skip generating a domain entry.

manut, 2018-11-30:
http://lists.linutronix.de/pipermail/elbe-devel/2018-November/001938.html

manut, 2018-12-05:
fixed by 886e9138cde7c75340f470ca0d3940b58fa9871a

closed 2018-12-05 by manut