Linutronix/elbe #188: elbe-testing doesn't work with local mirror

manut, 2018-11-16:
debmirror currently doesn't retrieve the SHA256 file for the debian installer.
This is an (uncommented) issue since 03/2016 [0].

However elbes debinstaller.py only verifies the SHA256 file. We should consider
to fix debmirror or at least use the MD5SUM file for now.

As a workaround
---
wget -O debian/dists/jessie/main/installer-amd64/current/images/SHA256SUMS \
http://ftp.de.debian.org/debian/dists/jessie/main/installer-amd64/current/images/SHA256SUMS
---
can be added to the end of a debmirror wrapper script.

[0] https://bugs.debian.org/cgi-bin/bugreport.cgi?bug=819490

jogness, 2018-11-20:
Using the suggested workaround did not work for me because the actual images are not in the mirror either. As per:

https://wiki.debian.org/DebianCustomCD

I am using a different workaround that works for me. I run the following command after running debmirror:

    lftp -c mirror \
        http://ftp.de.debian.org/debian/dists/jessie/main/installer-amd64/current/ \
        ${MIRROR_ROOT}/debian/dists/jessie/main/installer-amd64/current

manut, 2018-11-20:
I use the following debmirror command. That also mirrors the debian installer images:

```
debmirror --getcontents -p --method=rsync --no-check-gpg \
	--di-arch=amd64 \
	--di-dist=jessie \
	-a armhf,amd64,i386,arm64,powerpc \
	-d jessie,jessie-backports,stretch,stretch-backports,buster,sid \
	-s main,main/debian-installer \
	-h ftp.de.debian.org \
	debian
```

jogness, 2018-11-20:
I was not aware of the `--di-arch` and `--di-dist` options. Thanks for pointing that out. Now I can use the original workaround.

closed 2019-03-12 by manut

torbenh3, 2019-03-29:
It seems like elbe is pointing people to this link currently, because the jessie installer images
have been removed from the debian mirrors.
We are working on a solution.