Linutronix/elbe #205: elbe-devel cannot buld arm64 buster images

manut, 2019-02-08:
the following error is shown: for <suite>buster</suite> and <arch>arm64</arch>

I: Configuring systemd...
I: Base system installed successfully.
------------------------------------------------------------------------------

running cmd +chroot /var/cache/elbe/3429c606-8db9-430f-be94-a004bbf5d7fb/chroot dpkg --configure -a+
------------------------------------------------------------------------------
------------------------------------------------------------------------------

running cmd +chroot /var/cache/elbe/3429c606-8db9-430f-be94-a004bbf5d7fb/chroot debconf-set-selections < /var/cache/elbe/3429c606-8db9-430f-be94-a004bbf5d7fb/chroot/var/cache/elbe/preseed.txt+
------------------------------------------------------------------------------
------------------------------------------------------------------------------

Keys imported
Build failed !!!

Failed to build the Apt Cache.
Not all Mirror Index Files could be downloaded.
Probable cause might be:
- Problems with Internet Connection
- Broken mirrors

Message from apt:
Error Updating rpcaptcache: W:There is no public key available for the following key IDs:
8B48AD6246925553
Signed-By: 126C0D24BD8A2942CC7DF8AC7638D0442B90D010!
 16E90B3FDF65EDE3AA7F323C04EE7237B7D453EC!
 E1CF20DDFFE4B89E802658F1E0B11894F66AEC98
URI: gpgv:/var/lib/apt/lists/partial/ftp.de.debian.org__debian_dists_buster_InRelease
Filename: /var/lib/apt/lists/partial/ftp.de.debian.org__debian_dists_buster_InRelease, E:The method driver /usr/lib/apt/methods/bzip2 could not be found.

manut, 2019-02-19:
this should be fixed if initvm uses a current debian release and a current debootstrap

manut, 2019-04-16:
The issue is resolved within elbe > 4 that uses stretch as initvm

closed 2019-04-16 by manut

mrelyb, 2019-07-23:
The issue is still there. When I try to build a buster image with elbe 9 by issuing "elbe buildchroot test.xml -t test" in the vm created by "elbe initrd create" I get the following error msg:

```
Keys imported
/usr/bin/python2: relocation error: /lib/x86_64-linux-gnu/libnss_files.so.2: symbol __libc_readline_unlocked, version GLIBC_PRIVATE not defined in file libc.so.6 with link time reference
Traceback (most recent call last):
  File "/usr/bin/elbe", line 55, in <module>
    cmdmod.run_command(sys.argv[2:])
  File "/usr/lib/python2.7/dist-packages/elbepack/commands/buildchroot.py", line 104, in run_command
    opt.skip_pbuild)
  File "/usr/lib/python2.7/dist-packages/elbepack/elbeproject.py", line 533, in build
    self.install_packages(self.buildenv)
  File "/usr/lib/python2.7/dist-packages/elbepack/elbeproject.py", line 897, in install_packages
    raise AptCacheUpdateError(e)
elbepack.elbeproject.AptCacheUpdateError: Error Updating rpcaptcache: 
```

When I try to execute the following python script I get the same error:

```
#!/usr/bin/env python
# aptinstall.py

import os
import apt
import sys

os.chroot('test/chroot')
cache = apt.cache.Cache()
cache.update()
cache.open()
```

So probably elbe is using different versions of libc and libnss_files after chroot?

torbenh3, 2019-08-16:
what mrelyb is seeing, is a different issue (see #231, which i have created now)