Linutronix/elbe #206: elbe fails on signature check using Ubuntu 18.04

gspanetta, 2019-02-18:
When I pull the current version on my Ubuntu 18.04 host machine and try to build elbe, it fails after a few seconds with the output shown below.
Note that when I use the Debian binary image from debian.linutronix.de repository, everything works fine.

```
$ ./elbe initvm create examples/armhf-ti-beaglebone-black.xml 
Import microsoft.gpg: 
gpg: Schlüssel EB3E94ADBE1229CF: Öffentlicher Schlüssel "Microsoft (Release signing) <gpgsecurity@microsoft.com>" importiert
gpg: Anzahl insgesamt bearbeiteter Schlüssel: 1
gpg:               importiert: 1
Import ubuntu-keyring-2012-archive.gpg: 
gpg: key 3B4FE6ACC0B21F32: 3 Beglaubigungen wegen fehlender Schlüssel nicht geprüft
gpg: Schlüssel 3B4FE6ACC0B21F32: Öffentlicher Schlüssel "Ubuntu Archive Automatic Signing Key (2012) <ftpmaster@ubuntu.com>" importiert
gpg: Anzahl insgesamt bearbeiteter Schlüssel: 1
gpg:               importiert: 1
Import ubuntu-keyring-2018-archive.gpg: 
gpg: key 871920D1991BC93C: 1 Beglaubigung wegen fehlendem Schlüssel nicht geprüft
gpg: Schlüssel 871920D1991BC93C: Öffentlicher Schlüssel "Ubuntu Archive Automatic Signing Key (2018) <ftpmaster@ubuntu.com>" importiert
gpg: Anzahl insgesamt bearbeiteter Schlüssel: 1
gpg:               importiert: 1
Import ubuntu-keyring-2012-cdimage.gpg: 
gpg: key D94AA3F0EFE21092: 3 Beglaubigungen wegen fehlender Schlüssel nicht geprüft
gpg: Schlüssel D94AA3F0EFE21092: Öffentlicher Schlüssel "Ubuntu CD Image Automatic Signing Key (2012) <cdimage@ubuntu.com>" importiert
gpg: Anzahl insgesamt bearbeiteter Schlüssel: 1
gpg:               importiert: 1
--2019-02-18 15:55:14--  http://ftp.de.debian.org//debian/dists/jessie/Release
Auflösen des Hostnamens ftp.de.debian.org (ftp.de.debian.org) … 141.76.2.4
Verbindungsaufbau zu ftp.de.debian.org (ftp.de.debian.org)|141.76.2.4|:80 … verbunden.
HTTP-Anforderung gesendet, auf Antwort wird gewartet … 200 OK
Länge: 148222 (145K)
Wird in »/tmp/tmpZSbRIi/Release« gespeichert.

/tmp/tmpZSbRIi/Release               100%[=====================================================================>] 144,75K  --.-KB/s    in 0,1s    

2019-02-18 15:55:14 (1,30 MB/s) - »/tmp/tmpZSbRIi/Release« gespeichert [148222/148222]

Signature with unknown key: 8B48AD6246925553
Signature with unknown key: 7638D0442B90D010
Signature with unknown key: CBF8D6FD518E17E1
Failure to download kernel/initrd debian Package:

InvalidSignature Failed to verify Release file

Check Mirror configuration
'elbe init' Failed
Giving up

```

manut, 2019-02-19:
Do you use the debian binary packages also on the ubuntu host machine? (That is completely untested btw.)

I guess your are missing the debian repo keys on your machine.
sudo apt-key adv --keyserver keyserver.ubuntu.com --recv-keys 8B48AD6246925553
for all the missing should do the job.

gspanetta, 2019-02-19:
Yes, I added `deb http://debian.linutronix.de/elbe-testing buster main` to my sources.list in Ubuntu and installed elbe via apt. It works quite well.

Thank you for the hint, now it works.

closed 2019-02-19 by manut