Linutronix/elbe #209: libjack-dev causes held broken packages

jogness, 2019-02-27:
If libjack-dev is added as a package, the log.txt shows:

`Unable to correct problems libjack-dev`

A further investigation shows that the reason for this (the exception text) is:

`E:Unable to correct problems, you have held broken packages.`

A dependency of this package requires preseeding. I added this to my XML file:

```
<project>
...
    <preseed>
        <conf owner="jackd2" key="jackd/tweak_rt_limits" type="boolean" value="true"/>
    </preseed>
</project>
```

The package **is** installed in the rootfs and the preseeding **did** work. Both "dpkg --get-selections" and "dpkg -l" show that there are no partially installed or broken packages.

However, the libjack-dev package is **not** listed in the source.xml so something has gone wrong even though the rootfs seems to be fine.

jogness, 2019-02-27:
I found some more information:
The interesting package (dependency) is jackd2. In the project directory of elbe I see:

- in the chroot directory, jackd2 is installed
- in the target directory, jackd2 is **not** installed

How can a package installed in chroot disappear in target?

jogness, 2019-02-27:
I stumbled on some more information. At the very end of the log.txt, after elbe seems to do one last "apt-get update", I see the following (notice the removing and something special with the jack packages):

```
Hit ...
Hit http://debug.mirrors.debian.org stretch-debug/main amd64 Packages
Hit http://debug.mirrors.debian.org stretch-debug/contrib amd64 Packages
Hit http://debug.mirrors.debian.org stretch-debug/non-free amd64 Packages
Hit http://debug.mirrors.debian.org stretch-debug/contrib Translation-en
Hit http://debug.mirrors.debian.org stretch-debug/main Translation-en
Hit http://debug.mirrors.debian.org stretch-debug/non-free Translation-en
Get:1 http://10.100.61.6/debian/ stretch/main libjack-jackd2-0 amd64 1.9.10+20150825git1ed50c92~dfsg-5 [292 kB]
Get:2 http://10.100.61.6/debian/ stretch/main jackd2 amd64 1.9.10+20150825git1ed50c92~dfsg-5 [313 kB]
Get:3 http://10.100.61.6/debian/ stretch/main jackd all 5 [2020 B]
Get:4 http://10.100.61.6/debian/ stretch/main jack-stdio amd64 1.4-1+b2 [15.1 kB]
Get:5 http://10.100.61.6/debian/ stretch/main qjackctl amd64 0.4.4-1 [448 kB]
pmstatus:dpkg-exec:0:Running dpkg
pmstatus:libjack0:0:Removing libjack0 (amd64)
pmstatus:libjack0:3.44828:Preparing for removal of libjack0 (amd64)
pmstatus:libjack0:6.89655:Removing libjack0 (amd64)
pmstatus:libjack0:10.3448:Removed libjack0 (amd64)
pmstatus:libc-bin:10.3448:Running post-installation trigger libc-bin
pmstatus:dpkg-exec:10.3448:Running dpkg
pmstatus:libjack-jackd2-0:10.3448:Installing libjack-jackd2-0 (amd64)
pmstatus:libjack-jackd2-0:13.7931:Preparing libjack-jackd2-0 (amd64)
pmstatus:libjack-jackd2-0:17.2414:Unpacking libjack-jackd2-0 (amd64)
pmstatus:libjack-jackd2-0:20.6897:Preparing to configure libjack-jackd2-0 (amd64)
pmstatus:jackd2:20.6897:Installing jackd2 (amd64)
pmstatus:jackd2:24.1379:Preparing jackd2 (amd64)
pmstatus:jackd2:27.5862:Unpacking jackd2 (amd64)
pmstatus:jackd2:31.0345:Preparing to configure jackd2 (amd64)
pmstatus:jackd:31.0345:Installing jackd (amd64)
pmstatus:jackd:34.4828:Preparing jackd (amd64)
pmstatus:jackd:37.931:Unpacking jackd (amd64)
pmstatus:jackd:41.3793:Preparing to configure jackd (amd64)
pmstatus:jack-stdio:41.3793:Installing jack-stdio (amd64)
pmstatus:jack-stdio:44.8276:Preparing jack-stdio (amd64)
pmstatus:jack-stdio:48.2759:Unpacking jack-stdio (amd64)
pmstatus:jack-stdio:51.7241:Preparing to configure jack-stdio (amd64)
pmstatus:qjackctl:51.7241:Installing qjackctl (amd64)
pmstatus:qjackctl:55.1724:Preparing qjackctl (amd64)
pmstatus:qjackctl:58.6207:Unpacking qjackctl (amd64)
pmstatus:qjackctl:62.069:Preparing to configure qjackctl (amd64)
pmstatus:libc-bin:62.069:Running post-installation trigger libc-bin
pmstatus:man-db:62.069:Running post-installation trigger man-db
pmstatus:desktop-file-utils:62.069:Running post-installation trigger desktop-file-utils
pmstatus:mime-support:62.069:Running post-installation trigger mime-support
pmstatus:hicolor-icon-theme:62.069:Running post-installation trigger hicolor-icon-theme
pmstatus:dpkg-exec:62.069:Running dpkg
pmstatus:libjack-jackd2-0:62.069:Configuring libjack-jackd2-0 (amd64)
pmstatus:libjack-jackd2-0:65.5172:Configuring libjack-jackd2-0 (amd64)
pmstatus:libjack-jackd2-0:68.9655:Installed libjack-jackd2-0 (amd64)
pmstatus:jackd2:68.9655:Configuring jackd2 (amd64)
pmstatus:jackd2:72.4138:Configuring jackd2 (amd64)
pmstatus:jackd2:75.8621:Installed jackd2 (amd64)
pmstatus:jackd:75.8621:Configuring jackd (amd64)
pmstatus:jackd:79.3103:Configuring jackd (amd64)
pmstatus:jackd:82.7586:Installed jackd (amd64)
pmstatus:jack-stdio:82.7586:Configuring jack-stdio (amd64)
pmstatus:jack-stdio:86.2069:Configuring jack-stdio (amd64)
pmstatus:jack-stdio:89.6552:Installed jack-stdio (amd64)
pmstatus:qjackctl:89.6552:Configuring qjackctl (amd64)
pmstatus:qjackctl:93.1035:Configuring qjackctl (amd64)
pmstatus:qjackctl:96.5517:Installed qjackctl (amd64)
pmstatus:libc-bin:96.5517:Running post-installation trigger libc-bin
pmstatus:dpkg-exec:96.5517:Running dpkg
100% update finished
```

virraaj, 2022-07-15:
I tried building for armhf for buster, bullseye having "libjack-dev" included in package-list. And with elbe 14.4, I could not see any problems.

bgermann, 2022-07-18:
The target system is stretch. @jogness Do you still need this to be fixed?

jogness, 2022-07-18:
Since this bug was opened, I have switched to libjack-jackd2-dev. So I no longer need libjack-dev. However, since [virraaj](https://github.com/virraaj) says it is working now, this issue can be closed anyway.

closed 2022-07-18 by jogness