Linutronix/elbe #211: elbe4: debian-installer breaks at "Select and install software"

bgermann, 2019-04-09:
`elbe initvm create` aborts at "Select and install software" for Elbe v4. Yesterday I built one initvm successfully but later had the same issue that is reported by @jogness at https://github.com/Linutronix/elbe/issues/188. So maybe it was an update on an Elbe dependency yesterday (afternoon) that broke it.
The full message looks like:

```
Select and install software  ... 10%... 20%... 31%... 40%... 50%... 61%... 70%... 80%... 90%
Select and install software
---------------------------

!! ERROR: Installation step failed

An installation step failed. You can try to run the failing item again from the
menu, or skip it and choose something else. The failing step is: Select and 
install software
[Press enter to continue]
```

markusmoll, 2019-04-09:
I am experiencing the same problem.

bgermann, 2019-04-10:
The stretch-backports kernel that was missing and caused this issue (see mail thread) is now available again.

closed 2019-04-10 by bgermann