Linutronix/elbe #212: qemu: Unsupported syscall: 383

bgermann, 2019-04-14:
For an elbe armhf pbuilder building a package with man-db I hit https://bugs.debian.org/cgi-bin/bugreport.cgi?bug=891109. The process loops on "qemu: Unsupported syscall: 383". As almost every Debian package has a (transitive) build dependency on man-db this is going to hit other users as well. The bug is fixed for buster but is not available in stretch-backports.

manut, 2019-04-29:
I cannot reproduce this.

Can you describe how to reproduce this issue please.

bgermann, 2019-04-30:
I cannot reproduce anymore as well, so closing this.

bgermann, 2019-05-14:
This can be triggered with <buildtype>armhf</buildtype> ... <suite>buster</suite> ... <pbuilder><git revision='133c0bd417532663afc8cbaad852137dd34cdef9'>https://github.com/labgrid-project/labgrid.git</git></pbuilder>

bgermann, 2019-06-28:
I guess moving the initvm to buster will fix the problem with buster's newer qemu version.
As long as this is not done the example hook /usr/share/doc/pbuilder/examples/D80no-man-db-rebuild can be registered for elbe's pbuilder runs.

bgermann, 2019-07-02:
With a buster initvm this is fixed indeed.

bgermann, 2019-09-02:
Fixed by 303a5d63d1e6193

closed 2019-09-02 by bgermann