Linutronix/elbe #214: Docker build fails with "useradd: UID 0 is not unique"

QuickJack, 2019-04-16:
Steps to reproduce:
1. `cd contrib/dockerfile`

2. `sudo -E make build`

Result:

> Step 18/30 : RUN useradd -d /home/elbe -U -G kvm-elbe,libvirt -m -s /bin/bash -u 0 elbe
>  ---> Running in c35d7ce9f8bc
> useradd: UID 0 is not unique
> The command '/bin/sh -c useradd -d /home/elbe -U -G kvm-elbe,libvirt -m -s /bin/bash -u 0 elbe' returned a non-zero code: 4
> make: *** [build] Fehler 4

Solution:
The option -o needs to be added to useradd:
https://github.com/Linutronix/elbe/blob/82806d61012b08dca39e0b2de5a8429b5f98697d/contrib/dockerfile/Dockerfile.in#L63

manut, 2019-04-16:
The code should add the current user to the container. Because you're using sudo  this is ID 0 and already exists. You shouldn't need to run docker as root.

I don't like adding the -o option, but maybe a check in the Makefile for not running it as root would be nice.

QuickJack, 2019-04-16:
By default, the docker command can only be run by the root user. This could be changed but it's not the default configuration. Because of that, many users will run into this problem.

kindtek, 2022-11-04:
> Solution: The option -o needs to be added to useradd:
> 
> https://github.com/Linutronix/elbe/blob/82806d61012b08dca39e0b2de5a8429b5f98697d/contrib/dockerfile/Dockerfile.in#L63

Thank you so much for posting this about adding -o. You saved me countless hours and headaches

bgermann, 2023-05-26:
The container can run rootless if the user is able to use the /dev/kvm device. Please read the elbe documentation which groups the user has to be assigned to.

closed 2023-05-26 by bgermann