Linutronix/elbe #215: v3: elbe initvm fails

QuickJack, 2019-04-16:
The output log ends as follows:

2019-04-16 08:22:38 (112 MB/s) - ‘/tmp/tmptQz17J/Release’ saved [148222/148222]

> Debian Archive Automatic Signing Key <ftpmaster@debian.org> (A1BD8E9D78F7FE5C3E65D8AF8B48AD6246925553):
> VALID (Untrusted).
> Debian Archive Automatic Signing Key <ftpmaster@debian.org> (126C0D24BD8A2942CC7DF8AC7638D0442B90D010):
> VALID (Untrusted).
> Jessie Stable Release Key <debian-release@lists.debian.org> (75DDC3C4A499F1A18CB5F3C8CBF8D6FD518E17E1):
> VALID (Untrusted).
> --2019-04-16 08:22:38--  http://ftp.de.debian.org//debian/dists/jessie/main/installer-amd64/current/images/SHA256SUMS
> Connecting to 172.18.0.100:8080... connected.
> Proxy request sent, awaiting response... 404 Not Found
> 2019-04-16 08:22:38 ERROR 404: Not Found.
> 
> Failure to download kernel/initrd debian Package:
> 
> HashValidationFailed Failed to download http://ftp.de.debian.org//debian/dists/jessie/main/installer-amd64/current/images/SHA256SUMS
> 
> Check Mirror configuration
> If you use debmirror please read https://github.com/Linutronix/elbe/issues/188 on how to work around the issue
> 'elbe init' Failed
> Giving up
> elbe_build
> elbe_build

curl  http://ftp.de.debian.org//debian/dists/jessie/main/installer-amd64/current/images

> /SHA256SUMS
> <!DOCTYPE HTML PUBLIC "-//IETF//DTD HTML 2.0//EN">
> <html><head>
> <title>404 Not Found</title>
> </head><body>
> <h1>Not Found</h1>
> <p>The requested URL /debian/dists/jessie/main/installer-amd64/current/images/SHA256SUMS was not found on this server.</p>
> <hr>
> <address>Apache/2.4.25 (Debian) Server at ftp.de.debian.org Port 80</address>
> </body></html>

QuickJack, 2019-04-16:
I have just found out that these files do no longer exist for jessie but for stretch.

manut, 2019-04-16:
The issue is resolved within elbe 2.5 and elbe > 4.

http://lists.linutronix.de/pipermail/elbe-devel/2019-April/002169.html

closed 2019-04-16 by manut