Linutronix/elbe #217: elbe v6: elbe init failure

richardcochran, 2019-04-26:
Wanted to try shiny new v6, running on debian 9.8 stable.
Any ideas?
```
elbe init --nesting /usr/share/doc/elbe-doc/examples/elbe-init-big-machine.xml
...
--2019-04-26 07:18:10--  http://ftp.de.debian.org//debian/dists/stretch/main/installer-amd64/current/images/netboot/debian-installer/amd64/linux
Resolving ftp.de.debian.org (ftp.de.debian.org)... 141.76.2.4
Connecting to ftp.de.debian.org (ftp.de.debian.org)|141.76.2.4|:80... connected.
HTTP request sent, awaiting response... 200 OK
Length: 4241184 (4.0M)
Saving to: ‘/tmp/tmpVjBn_G/vmlinuz’

/tmp/tmpVjBn_G/vmlinuz                  100%[==============================================================================>]   4.04M  2.63MB/s    in 1.5s    

2019-04-26 07:18:12 (2.63 MB/s) - ‘/tmp/tmpVjBn_G/vmlinuz’ saved [4241184/4241184]

Traceback (most recent call last):
  File "/usr/lib/python2.7/dist-packages/elbepack/templates.py", line 28, in template
    **d)
  File "/usr/lib/python2.7/dist-packages/mako/template.py", line 462, in render
    return runtime._render(self, self.callable_, args, data)
  File "/usr/lib/python2.7/dist-packages/mako/runtime.py", line 838, in _render
    **_kwargs_for_callable(callable_, data))
  File "/usr/lib/python2.7/dist-packages/mako/runtime.py", line 873, in _render_context
    _exec_template(inherit, lclcontext, args=args, kwargs=kwargs)
  File "/usr/lib/python2.7/dist-packages/mako/runtime.py", line 899, in _exec_template
    callable_(context, *args, **kwargs)
  File "/usr/lib/python2.7/dist-packages/elbepack/init/libvirt.xml.mako", line 8, in render_body
    <%
  File "/usr/lib/python2.7/posixpath.py", line 70, in join
    elif path == '' or path.endswith('/'):
AttributeError: 'NoneType' object has no attribute 'endswith'

Traceback (most recent call last):
  File "/usr/bin/elbe", line 55, in <module>
    cmdmod.run_command(sys.argv[2:])
  File "/usr/lib/python2.7/dist-packages/elbepack/commands/init.py", line 218, in run_command
    init_template_dir, t), d, linebreak=True)
  File "/usr/lib/python2.7/dist-packages/elbepack/templates.py", line 38, in write_template
    outfile.write(template(fname, d, linebreak))
  File "/usr/lib/python2.7/dist-packages/elbepack/templates.py", line 28, in template
    **d)
  File "/usr/lib/python2.7/dist-packages/mako/template.py", line 462, in render
    return runtime._render(self, self.callable_, args, data)
  File "/usr/lib/python2.7/dist-packages/mako/runtime.py", line 838, in _render
    **_kwargs_for_callable(callable_, data))
  File "/usr/lib/python2.7/dist-packages/mako/runtime.py", line 873, in _render_context
    _exec_template(inherit, lclcontext, args=args, kwargs=kwargs)
  File "/usr/lib/python2.7/dist-packages/mako/runtime.py", line 899, in _exec_template
    callable_(context, *args, **kwargs)
  File "_usr_lib_python2_7_dist_packages_elbepack_init_libvirt_xml_mako", line 46, in render_body
  File "/usr/lib/python2.7/posixpath.py", line 70, in join
    elif path == '' or path.endswith('/'):
AttributeError: 'NoneType' object has no attribute 'endswith'
```

manut, 2019-04-26:
this is a bug, if --directory is not specified.

a fix is proposed here: http://lists.linutronix.de/pipermail/elbe-devel/2019-April/002326.html

richardcochran, 2019-04-26:
That fixes the issue for me, thanks!

commit 29065807 (2019-04-29): init: fix crash if --directory was not specified

this fixes #217 where opt.directory is accessed from
elbepack/init/libvirt.xml.mako
copied from https://github.com/Linutronix/elbe/issues/217
----------------------------------------------------------------------------------------------------------
Traceback (most recent call last):
  File "/usr/lib/python2.7/dist-packages/elbepack/templates.py", line 28, in template
    **d)
  File "/usr/lib/python2.7/dist-packages/mako/template.py", line 462, in render
    return runtime._render(self, self.callable_, args, data)
  File "/usr/lib/python2.7/dis [... cut]
closed 2019-04-29 by manut