Linutronix/elbe #218: install_version is broken

manut, 2019-05-06:
$ elbe control list_projects
elbe v6 is used in initvm, this is not compatible with elbe v7 that is used on this machine. Please install same versions of elbe in initvm and on your machine.
To install elbe v7 into the initvm use 'elbe control install_elbe_version'
$ elbe control install_elbe_version
elbe v6 is used in initvm, this is not compatible with elbe v7 that is used on this machine. Please install same versions of elbe in initvm and on your machine. To install elbe v7 into the initvm use 'elbe control install_elbe_version'
$ elbe control install_elbe_version --ignore-version-diff
elbe v6 is used in initvm, this is not compatible with elbe v7 that is used on this machine. Please install same versions of elbe in initvm and on your machine.
To install elbe v7 into the initvm use 'elbe control install_elbe_version'
Reading package lists...
Building dependency tree...
Reading state information...
E: Version '7*' for 'python-elbe-buildenv' was not found
E: Version '7*' for 'elbe-soap' was not found
E: Version '7*' for 'python-elbe-common' was not found
E: Version '7*' for 'elbe-daemon' was not found
E: Version '7*' for 'elbe-schema' was not found
E: Version '7*' for 'python-elbe-bin' was not found

Error: apt returns 100

hdengler, 2022-03-15:
There is a workaround for this bug:
- attach to initvm an login as root (elbe initvm attach)
- inside of the initvm:
  - update the package index (apt update)
  - update elbe packages (apt install python3-elbe-soap)
  - logout and detach

The versions should now be in sync.

bgermann, 2022-10-13:
Closing as claimed in the commit reference.

closed 2022-10-13 by bgermann