Linutronix/elbe #219: BBB example no init package

dedukun, 2019-05-09:
After following the example in the documentation to build the BBB image using `elbe initvm create armhf-ti-beaglebone-black.xml`,  two problems were found.

### Problem 1

The first was that elbe installed the latest kernel version, kernel 4.9.0-9 at the time of this issue, but it was configured in uEnv.txt to have version 4.9.0-3.

```
U-Boot SPL 2016.11+dfsg1-4 (Mar 27 2017 - 18:39:51)
Trying to boot from MMC1
MMC partition switch failed
*** Warning - MMC partition switch failed, using default environment

reading u-boot.img
reading u-boot.img

U-Boot 2016.11+dfsg1-4 (Mar 27 2017 - 18:39:51 +0000)

CPU  : AM335X-GP rev 2.0
I2C:   ready
DRAM:  512 MiB
MMC:   OMAP SD/MMC: 0, OMAP SD/MMC: 1
*** Warning - bad CRC, using default environment

<ethaddr> not set. Validating first E-fuse MAC
Net:   cpsw
Press SPACE to abort autoboot in 2 seconds
switch to partitions #0, OK
mmc0 is current device
SD/MMC found on device 0
reading boot.scr
** Unable to read file boot.scr **
reading uEnv.txt
255 bytes read in 6 ms (41 KiB/s)
Loaded env from uEnv.txt
Importing environment from mmc0 ...
Running uenvcmd ...
reading vmlinuz-4.9.0-3-armmp
** Unable to read file vmlinuz-4.9.0-3-armmp **
reading dtb-4.9.0-3-armmp
** Unable to read file dtb-4.9.0-3-armmp **
reading initrd.img-4.9.0-3-armmp
** Unable to read file initrd.img-4.9.0-3-armmp **
switch to partitions #0, OK
mmc0 is current device
Scanning mmc 0:1...
switch to partitions #0, OK
mmc0 is current device
SD/MMC found on device 0
switch to partitions #0, OK
mmc0 is current device
SD/MMC found on device 0
reading boot.scr
** Unable to read file boot.scr **
reading uEnv.txt
255 bytes read in 6 ms (41 KiB/s)
Loaded env from uEnv.txt
Importing environment from mmc0 ...
Running uenvcmd ...
reading vmlinuz-4.9.0-3-armmp
** Unable to read file vmlinuz-4.9.0-3-armmp **
reading dtb-4.9.0-3-armmp
** Unable to read file dtb-4.9.0-3-armmp **
reading initrd.img-4.9.0-3-armmp
** Unable to read file initrd.img-4.9.0-3-armmp **
** File not found /boot/zImage **
switch to partitions #0, OK
mmc1(part 0) is current device
Scanning mmc 1:1...
Error: FAT sector size mismatch (fs=26978, dev=512)
Error: FAT sector size mismatch (fs=26978, dev=512)
Error: FAT sector size mismatch (fs=26978, dev=512)
Error: FAT sector size mismatch (fs=26978, dev=512)
Error: FAT sector size mismatch (fs=26978, dev=512)
Error: FAT sector size mismatch (fs=26978, dev=512)
Error: FAT sector size mismatch (fs=26978, dev=512)
Error: FAT sector size mismatch (fs=26978, dev=512)
Error: FAT sector size mismatch (fs=26978, dev=512)
Error: FAT sector size mismatch (fs=26978, dev=512)
switch to partitions #0, OK
mmc1(part 0) is current device
SD/MMC found on device 1
switch to partitions #0, OK
mmc1(part 0) is current device
SD/MMC found on device 1
reading boot.scr
Error: FAT sector size mismatch (fs=26978, dev=512)
** Unable to read file boot.scr **
reading uEnv.txt
Error: FAT sector size mismatch (fs=26978, dev=512)
** Unable to read file uEnv.txt **
Running uenvcmd ...
reading vmlinuz-4.9.0-3-armmp
** Unable to read file vmlinuz-4.9.0-3-armmp **
reading dtb-4.9.0-3-armmp
** Unable to read file dtb-4.9.0-3-armmp **
reading initrd.img-4.9.0-3-armmp
** Unable to read file initrd.img-4.9.0-3-armmp **
** File not found /boot/zImage **
## Error: "bootcmd_nand0" not defined
```

This is what is inside the SD card after flashing.

```bash
total 23M
-rwxr-xr-x 1 root root 185K Apr 12 14:52 config-4.9.0-9-armmp
-rwxr-xr-x 1 root root  16M May  9 10:05 initrd.img-4.9.0-9-armmp
-rwxr-xr-x 1 root root  72K Mar 27  2017 MLO
-rwxr-xr-x 1 root root 2.9M Apr 12 14:52 System.map-4.9.0-9-armmp
-rwxr-xr-x 1 root root 358K Mar 27  2017 u-boot.img
-rwxr-xr-x 1 root root  255 May  9 10:05 uEnv.txt
-rwxr-xr-x 1 root root 3.6M Apr 12 14:52 vmlinuz-4.9.0-9-armmp
```

As a suggestion, so that this example works with no issues 'out of the box', maybe the version of the kernel installed should be explicit in the package list.

### Problem 2

The second problem found was that when the device is finishing booting, there's an error saying it doesn't has `init` in the system.

```
Begin: Loading essential drivers ... done.
Begin: Running /scripts/init-premount ... done.
Begin: Mounting root file system ... Begin: Running /scripts/local-top ... done.
Begin: Running /scripts/local-premount ... done.
Warning: fsck not present, so skipping root file system
[    4.037970] EXT4-fs (mmcblk0p2): mounting ext2 file system using the ext4 subsystem
[    4.051072] EXT4-fs (mmcblk0p2): mounted filesystem without journal. Opts: (null)
done.
Begin: Running /scripts/local-bottom ... done.
Begin: Running /scripts/init-bottom ... done.
run-init: /sbin/init: No such file or directory
Target filesystem doesn't have requested /sbin/init.
run-init: /sbin/init: No such file or directory
run-init: /etc/init: Permission denied
run-init: /bin/init: No such file or directory
``` 
### Fix 
The image works correctly after changing the package list in  `armhf-ti-beaglebone-black.xml` to look like this.

```xml
<pkg-list>
    <pkg>u-boot-omap</pkg>

    <!-- SOLVES PROBLEM 1 -->
    <pkg>linux-image-4.9.0-3-armmp</pkg>

    <pkg>openssh-server</pkg>
    <pkg>less</pkg>
    <pkg>bash</pkg>
    <pkg>vim-tiny</pkg>
    <pkg>wget</pkg>
    <pkg>ntpdate</pkg>
    <pkg>busybox</pkg>
    <!--
        explicit install of systemd, since debootstrap
        variant minbase has no default init system.
    -->
    <pkg>systemd</pkg>
    <!-- install a package from backports -->
    <pkg pin="stretch-backports">btrfs-progs</pkg>
    <!-- install the gpiotest built by pbuilder  -->
    <pkg>gpiotest</pkg>

    <!-- SOLVES PROBLEM 2  -->
    <pkg>init</pkg>
</pkg-list>
```

malagasonfire, 2019-05-09:
The following workaround works well using 4.9.0-9-armmp

<pre>
<pkg>linux-image-4.9.0-9-armmp</pkg>
</pre>

Change the finetunning section also to get the matching for dtb 

<pre>

	<finetuning>
			
			<cp path="/usr/lib/u-boot/am335x_boneblack/MLO">/boot/MLO</cp>
			
			<cp path="/usr/lib/u-boot/am335x_boneblack/u-boot.img">/boot/u-boot.img</cp>
			
			<cp path="/usr/lib/linux-image-4.9.0-9-armmp/am335x-boneblack.dtb">/boot/dtb-4.9.0-9-armmp</cp>
			
			<command>echo "uenvcmd=setenv bootargs 'console=ttyO0,115200 root=/dev/mmcblk0p2';load mmc 0:1 0x84000000 vmlinuz-4.9.0-9-armmp;load mmc 0:1 0x82000000 dtb-4.9.0-9-armmp;load mmc 0:1 0x88000000 initrd.img-4.9.0-9-armmp;bootz 0x84000000 0x88000000:\${filesize} 0x82000000" > /boot/uEnv.txt</command>
			
			<rm>/var/cache/apt/archives/*.deb</rm>
			<rm>/var/cache/apt/*.bin</rm>
			<rm>/var/lib/apt/lists/ftp*</rm>
			<artifact>usr/lib/u-boot/am335x_boneblack/MLO</artifact>
			<artifact>usr/lib/linux-image-4.9.0-9-armmp/am335x-boneblack.dtb</artifact>
		</finetuning>

</pre>

malagasonfire, 2019-05-09:
The following workaround works well using 4.9.0-9-armmp

<code>
<pkg>linux-image-4.9.0-9-armmp</pkg>
</code>

Change the finetunning section also to get the matching for dtb 

<code>

	<finetuning>
			
			<cp path="/usr/lib/u-boot/am335x_boneblack/MLO">/boot/MLO</cp>
			
			<cp path="/usr/lib/u-boot/am335x_boneblack/u-boot.img">/boot/u-boot.img</cp>
			
			<cp path="/usr/lib/linux-image-4.9.0-9-armmp/am335x-boneblack.dtb">/boot/dtb-4.9.0-9-armmp</cp>
			
			<command>echo "uenvcmd=setenv bootargs 'console=ttyO0,115200 root=/dev/mmcblk0p2';load mmc 0:1 0x84000000 vmlinuz-4.9.0-9-armmp;load mmc 0:1 0x82000000 dtb-4.9.0-9-armmp;load mmc 0:1 0x88000000 initrd.img-4.9.0-9-armmp;bootz 0x84000000 0x88000000:\${filesize} 0x82000000" > /boot/uEnv.txt</command>
			
			<rm>/var/cache/apt/archives/*.deb</rm>
			<rm>/var/cache/apt/*.bin</rm>
			<rm>/var/lib/apt/lists/ftp*</rm>
			<artifact>usr/lib/u-boot/am335x_boneblack/MLO</artifact>
			<artifact>usr/lib/linux-image-4.9.0-9-armmp/am335x-boneblack.dtb</artifact>
		</finetuning>

</code>

commit a6a9169a (2020-04-02): examples/beaglebone: Remove stretch versions

The beaglebone example builds without errors even though elbe cannot copy
some files in its finetuning section which contain a stretch kernel version.
Move existing files version independent using glob (*).

Fixes #219.

Reviewed-by: Torben Hohn <torben.hohn@linutronix.de>
closed 2020-04-02 by torbenh3