Linutronix/elbe #226: using pipe in finetuning command can fail with qemu

jogness, 2019-07-04:
Using pipes with finetuning commands, such as:

`<command>echo "hello" | tee /etc/mytest</command>`

will fail if building for a non-x86_64 architecture. It seems when qemu is started via binfmt, the PATH is not set for commands started via pipe. Changing the command to:

`<command>echo "hello" | /usr/bin/tee /etc/mytest</command>`

works.

To debug this, one could have a simple script to dump the environment variables:

    #!/bin/sh
    set

Then run this in finetuning:

`<command>echo "hello" | /path/to/myscript &gt; /etc/test.result</command>`

and investigate the variables/values listed in /etc/test.result on the target. It might be interesting to compare that output with:

`<command>/path/to/myscript &gt; /etc/test2.result</command>`

torbenh3, 2019-07-04:
maybe we do not have anything set in PATH.
echo is a shell builtin.

we are actually piping the command text to /bin/sh stdin.

oldiob, 2019-07-04:
I have done your test John.  The only difference between the two environnement is the PPID.

jogness, 2019-07-04:
> I have done your test John. The only difference between the two environnement is the PPID.

And you are using armhf for the target? The effect does not happen if the target is x86_64.

oldiob, 2019-07-04:
Same thing with armhf.

Tested with x86_64-docker.xml, just change the buildtype to armhf and add

~~~xml
<finetuning>
      <file mode="700" dst="/bin/myscript">           
         #!/bin/sh                                     
         set                                           
     </file>
    <command>echo "hello" | /bin/myscript &gt; /etc/test.result</command>
    <command>/bin/myscript &gt; /etc/test2.result</command>
</finetuning>
~~~

Outputs of test.result and test2.result (only diff is the PPID):
~~~
DAEMON_ARGS='daemon --port=7588 --soap'
DEBIAN_FRONTEND='noninteractive'
DEBONF_NONINTERACTIVE_SEEN='true'
IFS='
'
INVOCATION_ID='0976d381aa51427aa6b32ce23b89ae74'
JOURNAL_STREAM='9:1860'
LANG='C'
LANGUAGE='C'
LC_ALL='C'
OPTIND='1'
PATH='/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin'
PPID='8524'
PS1='# '
PS2='> '
PS4='+ '
PWD='/'
XML_CATALOG_FILES='/var/cache/elbe/devel/schema/catalog.xml'
http_proxy=''
https_proxy=''
no_proxy=''
~~~

oldiob, 2019-07-26:
```
If you cross-compile busybox from the git source,
with the same version that stretch is using (1.22.1), and you copy this version to your
target the bug is fixed!

Here's the step to follow:

       1. Have a broken busybox for ARM on your initvm

       2. Clone the git repo: <git://busybox.net/busybox.git> and
          checkout at tag 1_22_1

       3. make menuconfig with the following options:

          - [*] Build BusyBox as a static binary (no shared libs)
          - (arm-linux-gnueabi-) Cross Compiler prefix
          - (-I/usr/arm-linux-gnueabi/include) Additional CFLAGS

          and then compile.

       4. Execute the following script where $1 is the target fs:
       ---------------------------------------------------------------
       #!/bin/sh
       mkdir -p $1/bin $1/usr/bin $1/sys $1/proc $1/dev
       cp /usr/bin/qemu-arm-static $1/usr/bin
       mount -t proc none $1/proc
       mount -t sysfs none $1/sys
       mount -o bind /dev $1/dev
       mount -o bind /dev/pts $1/dev/pts
       ---------------------------------------------------------------

       5. Make symbolic links for different utilities ..
       ---------------------------------------------------------------
       cd my-target/bin
       ln -s /bin/busybox sh
       cd my-target/usr/bin
       ln -s /bin/busybox tee
       ---------------------------------------------------------------

       6. Copy a broken busybox to my-target/bin/busybox

       7. Chroot into the target and test

       8. Repeat 6. and 7. with the busybox you compiled in 3.

Here's something even more interesting.  Use the broken busybox and
execute a shell.  Then, do the following:
----------------------------------------------------------------------
# tee -h
----------------------------------------------------------------------

Here's the shorten output:
----------------------------------------------------------------------
usage: qemu-arm [options] program [arguments...]
Linux CPU emulator (compiled for arm emulation)
----------------------------------------------------------------------

Finally, here's a sample of strace of the previous command:
----------------------------------------------------------------------
...
[pid 17988] execve("/proc/self/exe", ["tee", "-h"], [/* 15 vars */] <unfinished ...>
...
[pid 17988] readlink("/proc/self/exe", "/usr/bin/qemu-arm-static", 4096) = 24
...
[pid 17988] access("/etc/ld.so.nohwcap", F_OK) = -1 ENOENT (No such file or directory)
...
[pid 17988] clone(child_stack=0x7fedceb6ddf0, flags=CLONE_VM|CLONE_FS|CLONE_FILES|CLONE_SIGHAND|CLONE_THREAD|CLONE_SYSVSEM|CLONE_SETTLS|
CLONE_PARENT_SETTID|CLONE_CHILD_CLEARTID, parent_tidptr=0x7fedceb6e9d0,
tls=0x7fedceb6e700, child_tidptr=0x7fedceb6e9d0) = 17990
...
[pid 17988] write(1, "usage: qemu-arm [options] progra"..., 49) = 49
[pid 17988] write(1, "Linux CPU emulator (compiled for"..., 48) = 48
...
----------------------------------------------------------------------

Now, I think the problem is that we normally have
'/proc/self/exe -> /bin/busybox' and thus it receives the argvs
{"tee", "-a"} which is OK. But here we probably have
'/proc/self/exe -> /usr/bin/qemu-arm-static'.

This is because of the BUSYBOX_EXEC_PATH configuration (see Config.in) that default to /proc/self/exec:
> /proc/self/exe always points to the currently running executable
which is not true in qemu-static!

As for why this is not happening when compiling from source, I have no idea.
I've tried compiling with different configurations, but it always works when compiling
from source.
```

oldiob, 2019-07-29:
Note that doing the following also work.  The previous comment detailed how the bug was discover.

~~~bash
qemu-arm-static /path-to/my-broken-busybox sh
~~~

bene42, 2023-10-25:
More info needed. Please reopen if nessesary.