Linutronix/elbe #227: Cannot change resolv.conf in finetuning command

matlec, 2019-07-17:
### What I did
We use systemd-resolved as resolver in our image. systemd-resolved creates a resolv.conf file at /run/systemd/resolve/ at runtime which must be symlinked to /etc/resolv.conf. I tried to create a symlink to this file with a finetuning command:
`<command>ln -sf /run/systemd/resolve/resolv.conf /etc/resolv.conf</command>`

### What I expected
I expected the target image to contain a symlink /etc/resolv.conf that points to /run/systemd/resolve/resolv.conf

### What I got instead
The target image contains a copy if the initvm's /etc/resolv.conf:
`nameserver 10.0.2.3`

### Additional notes
The problem is caused by ChRootFilesystem which saves a copy of /etc/resolv.conf (which is a copy of the initvm's resolv.conf) when entering the chroot environment and restores the file when leaving the environment. Leaving the chroot environment reverts the changes made by the finetuning command.

Currently we use the following work-around which relies on implementation details of Elbe:
`<command>rm /etc/resolv.conf.orig &amp;&amp; ln -sf /run/systemd/resolve/resolv.conf /etc/resolv.conf.orig</command>`

Related issue: #58

oldiob, 2019-07-18:
This is indeed due to the excursion in the ChRootFilesystem context manager.

I've made a patch that could fix that if you use the 'ln' tag instead of 'command'.  

Thus:
~~~xml
<ln path="/run/systemd/resolve/resolv.conf">/etc/resolv.conf</ln>
~~~

I can CC your email to the patch if you want to test it.  Please contact me at dion@linutronix.de if you would rather not share your email on GitHub.

matlec, 2019-07-19:
Thank you, I just sent you an email

torbenh3, 2019-08-28:
should be fixed with facdca7fe1e3fae323bc3fa3a62b5f576c85bcb9 and 1cdbf3212a8d95ac19d43e3fcf238b685876664f

closed 2019-08-28 by torbenh3